stm32duino/Arduino_Core_STM32 #423: compilation error for the processor STM32H7

gauix4, 2019-02-01:
Hello I'm having trouble compiling for a processor STM32H7
if anyone has a little idea it would be great thanks in advance :) 

arm-none-eabi-g++: error: unrecognized argument in option '-mcpu=cortex-m7'
arm-none-eabi-g++: note: valid arguments to '-mcpu=' are: arm1020e arm1020t arm1022e arm1026ej-s arm10e arm10tdmi arm1136j-s arm1136jf-s arm1156t2-s arm1156t2f-s arm1176jz-s arm1176jzf-s arm2 arm250 arm3 arm6 arm60 arm600 arm610 arm620 arm7 arm70 arm700 arm700i arm710 arm7100 arm710c arm710t arm720 arm720t arm740t arm7500 arm7500fe arm7d arm7di arm7dm arm7dmi arm7m arm7tdmi arm7tdmi-s arm8 arm810 arm9 arm920 arm920t arm922t arm926ej-s arm940t arm946e-s arm966e-s arm968e-s arm9e arm9tdmi cortex-a15 cortex-a5 cortex-a7 cortex-a8 cortex-a9 cortex-m0 cortex-m0plus cortex-m1 cortex-m3 cortex-m4 cortex-r4 cortex-r4f cortex-r5 cortex-r7 ep9312 fa526 fa606te fa626 fa626te fa726te fmp626 generic-armv7-a iwmmxt iwmmxt2 marvell-pj4 mpcore mpcorenovfp native strongarm strongarm110 strongarm1100 strongarm1110 xscale
arm-none-eabi-g++: error: unrecognized argument in option '-mfpu=fpv5-sp-d16'
arm-none-eabi-g++: note: valid arguments to '-mfpu=' are: crypto-neon-fp-armv8 fp-armv8 fpv4-sp-d16 neon neon-fp-armv8 neon-fp16 neon-vfpv4 vfp vfp3 vfpv3 vfpv3-d16 vfpv3-d16-fp16 vfpv3-fp16 vfpv3xd vfpv3xd-fp16 vfpv4 vfpv4-d16
arm-none-eabi-g++: error: unrecognized command line option '-std=gnu++14'
exit status 1
Erreur de compilation pour la carte Nucleo-H743ZI(400M)[SerialUART3]

fpistm, 2019-02-10:
Any update? Else I will close this as this is only issue with arm gcc used by end user.

gauix4, 2019-02-10:
thanks for your answer sorry i did not have time to start bending over it so i just try again now and i have a new problem

gauix4, 2019-02-10:
I am on Mac OS
Arduino : 1.8.8 (Mac OS X), Carte : "Nucleo-32, Nucleo L412KB, Mass Storage, Enabled with generic Serial, None, Smallest (-Os default), Newlib Nano (default)"

Les options de compilation ont été modifiées, tout sera recompilé
In file included from /Users/simon/Library/Arduino15/packages/STM32/hardware/stm32/1.4.0/system/Drivers/CMSIS/Device/ST/STM32L4xx/Include/stm32l4xx.h:245:0,
                 from /Users/simon/Library/Arduino15/packages/STM32/hardware/stm32/1.4.0/cores/arduino/stm32/stm32_def.h:24,
                 from /Users/simon/Library/Arduino15/packages/STM32/hardware/stm32/1.4.0/cores/arduino/stm32/clock.h:43,
                 from /Users/simon/Library/Arduino15/packages/STM32/hardware/stm32/1.4.0/cores/arduino/wiring_time.h:23,
                 from /Users/simon/Library/Arduino15/packages/STM32/hardware/stm32/1.4.0/cores/arduino/wiring.h:38,
                 from /Users/simon/Library/Arduino15/packages/STM32/hardware/stm32/1.4.0/cores/arduino/Arduino.h:32,
                 from sketch/sketch_feb11a.ino.cpp:1:
/Users/simon/Library/Arduino15/packages/STM32/hardware/stm32/1.4.0/system/Drivers/STM32L4xx_HAL_Driver/Inc/stm32l4xx_hal.h:46:32: fatal error: stm32l4xx_hal_conf.h: No such file or directory
 #include "stm32l4xx_hal_conf.h"

                                ^
compilation terminated.
exit status 1
Erreur de compilation pour la carte Nucleo-32

Ce rapport pourrait être plus détaillé avec
l'option "Afficher les résultats détaillés de la compilation"
activée dans Fichier -> Préférences.

fpistm, 2019-02-11:
Hi @gauix4 
It seems you have an environment installation issue.
Don't know exactly which one but not linked to the core itself but how you deal with core installation in Arduino IDE.
the stm32l4xx_hal_conf.h is in the variant selected. Here the NUCLEO_L412KB.
So, I guess your environnnment is not proper.
Start from a fresh install and use the board manager.

closed 2019-02-13 by fpistm