stm32duino/Arduino_Core_STM32 #425: Support for Nucleo STM32L452

tekk, 2019-02-01:
Hello guys,
Thanks for the amazing work.
I need support for Nucleo STM32L452.
Would that be possible?
Thank you.

fpistm, 2019-02-01:
Hi @tekk 
Have you consider adding it following:
https://github.com/stm32duino/wiki/wiki/Add-a-new-variant-(board)

fpistm, 2019-02-11:
@tekk 
FYI, I've been able to do the variant in //.
See #434

tekk, 2019-02-12:
Thank you! That's really awesome! Will try it as soon as I'll have some spare time. Thanks again. 🙂👍

fpistm, 2019-02-12:
Welcome.
Note, that I've also test the variant as I've got the board.

closed 2019-02-12 by fpistm

tekk, 2019-02-13:
That's good. Is there any possibility that the STM32Duino core would also work for this board apart from standard HAL libraries? Thanks.

fpistm, 2019-02-13:
I do not understand your question about working apart from HAL ?

tekk, 2019-02-19:
Sorry, I'm not native english speaker. I'm asking if this board could work with STM32Duino. But I think you're not the right person to ask, STM32Duino guys would need to support it, right?

fpistm, 2019-02-19:
Ok I think I understand your questions.
Roger's core (which does not use HAL) doesn't support L4 series. So no.
The only core supported it is this one.

tekk, 2019-02-19:
Nevermind. I will stick with the HAL layer. Thank you for your effort.

PR #434 (2019-02-12): [Nucleo L452RE] Add support
Fix #425
commit 0048ffdf (2019-02-12): [Nucleo L452RE] Add support

Fix #425