stm32duino/Arduino_Core_STM32 #428: EEPROM is erased on uploading new sketch

arihantdaga, 2019-02-08:
**Is your feature request/improvement related to a problem? Please describe.**
EEPROM is erased on uploading new sketch. 

**Describe the solution you'd like**
Can we give an option in upload menu for users to chose to clear only sketch or entire flash.

AnHardt, 2019-02-08:
The STM32 processors non~t have native EEPROM. EEPROM is simulated in flash, where also the program resides. The usual location of the simulated EEPROM in flash is at its end.
https://github.com/stm32duino/Arduino_Core_STM32/blob/0c6eccbb4e83a270be27656db31897f0dd71698a/cores/arduino/stm32/stm32_eeprom.c#L120-L135
That means, **if your program is to large it will erase the EEPROM**. Likely redefining a smaller FLASH_PAGE_SIZE will not work.
https://github.com/stm32duino/Arduino_Core_STM32/blob/5cc5415ca2e7c9906bfa29aee923a7f0d90c5473/cores/arduino/stm32/stm32_eeprom.h#L48-L58

arihantdaga, 2019-02-08:
@AnHardt I understand that, but what if sketch size is very well limited within the bounds and not reaching till the last sector, Lets say there is still 20kb space in flash and EEPROM is only last 1kb. Then we need not delete the EEPROM Content. If that can be given as an option to users in Arduino ide Menu, i think it'll be great. 
Currently Even if there is space left in Flash its deleting entire flash before uploading the code. 
`Sketch uses 60116 bytes (91%) of program storage space. Maximum is 65536 bytes.`
I tested with even smaller sketch also.

BennehBoy, 2019-02-09:
Assuming you are on windows, you will need to edit stlink_upload.bat in the tools folder.

You need to change the line:
`echo "" | stlink\ST-LINK_CLI.exe -c SWD -P %str% 0x8000000 -Rst -Run`

to read:
`echo "" | stlink\ST-LINK_CLI.exe -c SWD -P %str% 0x8000000 ske -Rst -Run`

The 'ske' parameter for the -P switch skips flash erase.

fpistm, 2019-02-21:
Probably could be added as a request for #445.

arihantdaga, 2019-03-16:
Cool thank you @BennehBoy . Its been long , but i finally went with this-
stm32flash cli support deleting only specific number of pages, 
so i modified on macOs my serial_upload file to this- 
```$(dirname $0)/stm32flash/stm32flash -g 0x8000000 -e 124 -b 115200 -w "$4" /dev/"$1"```

Just added -e 124 to erase only first 124 pages(Note that i am using bluepill with 128kb flash) and emulated eeprom is in last sector. 

But as suggested by @fpistm maybe in #445 we can give an option in arduino ide to select what to erase(Only Sketch or Sketch plus EEPROM).

fpistm, 2019-05-24:
Since #514, by default there is no full erase (which is also required to not erase bootloader if any)
As I do not want add again an extra menu, user can use `platform.local.txt` and add:
`tools.stm32CubeProg.upload.pattern="{path}/{cmd}" 1{upload.protocol} "{build.path}/{build.project_name}.bin" {upload.options}`

This is the same line except the '1' prefixing `{upload.protocol} `, this enable the full erase.

See https://github.com/stm32duino/Arduino_Tools/commit/00caaeb23de25b73cb03ae738043922fd8cad42d

closed 2019-05-24 by fpistm