stm32duino/Arduino_Core_STM32 #429: SPI DMA support

thomasfla, 2019-02-08:
SPI transfer is quite slow with a delay between each byte of about   
**1.4us** when calling `SPI.transfer(uint8* buff, int N); ` with a 9Mhz SPI clock. 

This delay is more than the duration of a byte transfer at this speed..

I think using DMA should solve this issue, but the current SPI library doesn't allow it.

Any chance this could be added to the todo features?

fpistm, 2019-02-10:
They are already some improvement requests for SPI.
DMA could be one of this. Anyway I will not have time soon to do this. Any help are welcome

VictorLamoine, 2019-02-18:
This should help you get started: http://stm32duino.com/viewtopic.php?f=14&t=3527&p=47009#p47009

fpistm, 2019-02-19:
Close this one to track all SPI request on the same issue #257.
Mark it as duplicated.

closed 2019-02-19 by fpistm