stm32duino/Arduino_Core_STM32 #430: MSVCR100.dll missing for ST-LINK_CLI.exe

indrek-tw, 2019-02-09:
Hi!
I was having trouble with uploading code to my Nucleo board after installing Arduino 1.8.8 and core STM32 board package on clean machine.
After digging here and there I double clicked on AppData\Local\Arduino15\packages\STM32\tools\STM32Tools\1.2.0\tools\win\stlink\ST-LINK_CLI.exe. And voila - it said that MSVCR100.dll is missing on computer. After installing Microsoft Visual C++ 2010 Redistributable everything was OK.
So I wonder why it wasnt resolved automatically during the install of core STM32 package? It was a bit hard to figure out because when running ST-LINK_CLI.exe from cmd line without having this DLL nothing actually happens - I mean its absolutely silent (I suppose the return code indicates the error).

fpistm, 2019-02-10:
> So I wonder why it wasnt resolved automatically during the install of core STM32 package?

Arduino install is only archive management. There is no check of what is install on the host or anything else.

Anyway, tools will be updated soon to use the STM32CubeProgrammer which will have to be install by end user because it could not be provided as a arduino package (at least to hard to maintain).

In the meantime, I will add a not in the wiki about that.

closed 2019-02-11 by fpistm

indrek-tw, 2019-02-11:
Thanks, hopefully it will help somebody. One clarification: when trying to upload then Arduino IDE will finish with "An error occurred while uploading the sketch". I was a bit unclear in my report, sorry. When running the command manually then this is silent (just exit code is -1), ie. this C:\Users\xyz\AppData\Local\Arduino15\packages\STM32\tools\STM32Tools\1.2.0/tools/win/stlink_upload.bat COM5 {upload.altID} {upload.usbID} C:\Users\xyz\AppData\Local\Temp\arduino_build_422616/userButton.ino.bin

padcom, 2021-04-22:
@ikruusa Any idea where to download the vc2010 redist package in 2021? I've been looking all over the net and all links point to a 404 at ms download center. That is sooooooo anoying!

ikruusa, 2021-04-22:
Haven't been hunting for this package long time. I dont have any better
pointer than what Google offers.
Indrek

padcom, 2021-04-23:
This is how I solved the problem - maybe someone will find it useful:

- switch to Linux
- use https://github.com/stlink-org/stlink instead

My original problem was that I tried to use the configurator from STorM32 and flash the controller from within that tool. I used Windows 10 because that's the only windows-like OS that I have had at my disposal in that particular situation and the STorM32 configurator is not available for anything else than Windows. It didn't use to be much of a problem before (just download the redist package, and you're done). But now that the download is literally GONE from MS downloads it is next to impossible to use apps like the `st-link_cli.exe` anymore.

Installing the `stlink`  from source took literally under a minute, everything compiled right off the bat so a simple `make && sudo make install` was all I needed to do to get it working.

fpistm, 2021-04-23:
I guess you used this core https://github.com/rogerclarkmelbourne/Arduino_STM32
which continue to use this exe while this core does not anymore.

padcom, 2021-04-23:
Dunno.. it came with the STorM32 configurator. God only knows what's in the package...

Kees-van-der-Oord, 2021-05-11:
I got st-link working after installing the vc 2010 distributables (x64 and x32):
https://www.microsoft.com/en-us/download/details.aspx?id=26999
Before that I installed the distributables of 2019, 2017, 2015 and 2012, so I don't know if those of 2010 are sufficient or you need some other as well.