stm32duino/Arduino_Core_STM32 #431: Possible I2C bug - no devices recognized

Nufflee, 2019-02-10:
I have a LSM9DS1 sensor connected to my STM32F407VG-DISC1 board but the I2C scanner example doesn't detect it. The wiring is correct because I can use it outside Arduino (STM32CubeMX with lower level libraries). 
I have it connected on PB9 (I2C1 SDA) and PB8 (I2C1 SCL) so I modified the sketch a bit ([hastebin of the modified sketch](https://hastebin.com/esiyaqetaj.cpp)) but I've also tried using different pins.
 
I would've reported this issue on the forums but registration seems to be disabled so I can't.

fpistm, 2019-02-10:
Have you pull up with a resistor the i2c lines?

Nufflee, 2019-02-10:
@fpistm  Oh, will try that out! And the LSM9DS1 has internal 10k pullups.

Nufflee, 2019-02-10:
Yup, that fixes it, thanks!

closed 2019-02-10 by Nufflee