stm32duino/Arduino_Core_STM32 #436: bugs in USBSerial & usbd_conf.c

stefaandesmet2003, 2019-02-12:
- Arduino IDE version: 1.8.8
 - STM32 core version: github latest
- OS: linux ubuntu
- board: bluepill
The 2 main issues i have :
1) USBSerial gives errors because the pma buffers for the CDC_OUT_EP endpoint is not correctly initialized in usbd_conf.c. Errors I have : hard fault when usb is plugged out and back in; dtr/rts not set correctly, serial monitor in arduino plugin in vscode corrupts the pma buffer (still don't understand why.., because cat /dev/ttyACM0 and arduino serial monitor work ok)
This code solves the issue :
`
// defines have to align with usbdc_cdc.h
#define CDC_IN_EP                                   0x81U  /* EP1 for data IN */
#define CDC_OUT_EP                                  0x01U  /* EP1 for data OUT */
#define CDC_CMD_EP                                  0x82U  /* EP2 for CDC commands */
  HAL_PCDEx_PMAConfig(pdev->pData, 0x00, PCD_SNG_BUF, 0x40);
  HAL_PCDEx_PMAConfig(pdev->pData, 0x80, PCD_SNG_BUF, 0x80);
  HAL_PCDEx_PMAConfig(pdev->pData, CDC_IN_EP, PCD_SNG_BUF, 0xC0);
  HAL_PCDEx_PMAConfig(pdev->pData, CDC_OUT_EP, PCD_SNG_BUF, 0x110);
  HAL_PCDEx_PMAConfig(pdev->pData, CDC_CMD_EP, PCD_SNG_BUF, 0x100);
`
2) CDC_Flush() can be called without CDC driver initialized, resulting in address 0 reference in USBD_CDC_SetTxBuffer.
This happens for instance when doing UsbSerial.print(...) with usb disconnected until the internal buffer fills up and code calls CDC_Flush(), or when calling UsbSerial.flush()

3) what's the advantage of using a timer for USBSerial? It can be implemented without by flushing to PMA on every write. It looks like unnecessary use of a timer peripheral (implementation in mbed or Roger Clark's repo)

stefaandesmet2003, 2019-02-12:
unfortunately no ! will definitely try.

stefaandesmet2003, 2019-02-12:
nice work! that branch solves my problems. The cdc_queue works well, blocks the write function once the queue is full; i guess that's intended behaviour.
issue closed :)
Thanks,
Stefaan

fpistm, 2019-02-12:
Welcome.
All congratulations go to @makarenya ;)
I've only reformat PR.

closed 2019-02-12 by fpistm