stm32duino/Arduino_Core_STM32 #438: STM32G0 support

pavkriz, 2019-02-13:
**Board description**
 - Name: NUCLEO-G071RB

When can we expect the support for STM32G0 series MCUs?

They are quite hacker-friendly thanks to single power rail (less coupling capacitors needed, simpler PCB), accurate internal RC oscilator (Xtal may not be needed), etc.

BennehBoy, 2019-02-13:
Maybe try add it yourself, there's a tutorial in the wiki....

BennehBoy, 2019-02-13:
Just looked in the tools and the pinmap source is there - STM32G071R(6-8-B)Tx  so should be doable...

fpistm, 2019-02-14:
Currenlty, G0 is not supported as Cube FW is not integrated.
This could be done as for WB.

BennehBoy, 2019-02-14:
@fpistm So for anyone ambitious I guess it's a case of following the steps in this PR -> https://github.com/stm32duino/Arduino_Core_STM32/pull/392 but for G0.

fpistm, 2019-02-14:
Right 👍

fpistm, 2019-02-26:
Hi  @pavkriz 
I've submitted the G0 support including Nucleo G071RB board variant: #459 
Note that it has not been tested as I'm waiting the board.

PR #459 (2019-03-06): Add STM32 G0 support and Nucleo-G071RB variant
This PR add the STM32 G0 support based on STM32 G0 HAL and CMSIS v1.1.0.
This also add the support of [Nucleo-G071RB](https://www.st.com/en/evaluation-tools/nucleo-g071rb.html)

Fix #438

# Restrictions: 
 * Upload only available thanks the Mass Storage. STLink is not updated to support G0 target, anyway it is possible to use the STM32CubeProgrammer to upload manually the binary.
 * Workaround for mass storage upload: Current ST-Link firmware limits the RAM size to 32k while 36k is
    available if parity check is disabled. This could be revert wh [... cut]