stm32duino/Arduino_Core_STM32 #455: Hard Fault in SPI::transfer

david-sawatzke, 2019-02-22:
When reporting any issue, please try to provide all relevant information:

**Describe the bug**
When running the [X-NUCLEO-IHM02A1](https://github.com/stm32duino/X-NUCLEO-IHM02A1) example, it gets stuck in the first init call. When attaching with the debug probe (via openocd & gdb), the resulting backtrace indicates a fault: 
```
#0  0x08005f18 in WWDG_IRQHandler ()
#1  <signal handler called>
#2  0x08004b10 in HAL_SPI_TransmitReceive ()
#3  0x08005eb6 in spi_transfer ()
#4  0x080017e0 in SPIClass::transfer(unsigned char, void*, void*, unsigned int, SPITransferMode) ()
#5  0x08001ec4 in L6470::L6470_DaisyChainCommand(unsigned char*, unsigned char*) ()
#6  0x08002476 in L6470::L6470_Config(void*) ()
#7  0x08002cce in L6470::init(void*) ()
#8  0x08002f5a in XNucleoIHM02A1::init_L6470_0() ()
#9  0x08002fa2 in XNucleoIHM02A1::init() ()
#10 0x08003060 in XNucleoIHM02A1::XNucleoIHM02A1(L6470_init_t*, L6470_init_t*, unsigned char, unsigned char, unsigned char, unsigned char, SPIClass*) ()
#11 0x08000f4c in setup ()
#12 0x080034a6 in main ()
```

Looking at the ICSR indicates a hard fault:
```
(gdb) x 0xE000ED04
0xe000ed04:	0x0400f803
```
Looking at the fault registers 
```
(gdb) x 0xE000ED2C
0xe000ed2c:	0x40000000
(gdb) x 0xE000ED28
0xe000ed28:	0x00008200
(gdb) x 0xE000ED38
0xe000ed38:	0xb294ee48
```
Indicates the faulting instruction is at 0xb294ee48 (no idea how).
This is the accompanying  arduino build folder
[arduino_build_624120.zip](https://github.com/stm32duino/Arduino_Core_STM32/files/2896064/arduino_build_624120.zip).

**To Reproduce**
Load the sample code onto the nucleo with the attached board and wait. After a while, attach via the debug probe and look at the backtrace.

**Expected behaviour**

Not hanging as an spi master (and not going in an exception).

**Desktop:**
 - OS: Linux
 - Arduino IDE version: 1.8.8
 - STM32 core version: 1.4.0
 - Tools menu settings if not the default: None
 - Upload method: Stlink

**Board :**
 - Name: Nucleo F429ZI
 - Hardware Revision: Rev B
 - Extra hardware used if any: X-Nucleo IHM02A1

fpistm, 2019-02-25:
Hi, thanks for the report.
@cparata an idea on this?

cparata, 2019-02-25:
Hi,
I guess the issue is due to the fact that on F429ZI in D3 (PE13) we do not have the SPI1 clock. So you are obliged to use D13 instead D3 as SPI clock. In order to do that you need to unsolder SB34 and solder SB12 on X-NUCLEO-IHM02A1 and then initialize the SPI instance in the sketch in this way:

dev_spi = new SPIClass(D11, D12, D13);

I hope that this tip can help you to solve your issue.
Best Regards,
Carlo

fpistm, 2019-02-25:
Thanks @cparata! Nice shot ;)
As the SCLK is not a good pin, it return.
https://github.com/stm32duino/Arduino_Core_STM32/blob/54478a1e6dad650ae14a08ec657419a01b7d4b6c/cores/arduino/stm32/spi_com.c#L151

david-sawatzke, 2019-02-25:
@cparata Ah, thanks, didn't catch that. Works fine now

@fpistm 
Wouldn't it be a better idea to just loop after an error condition like this? 

I'll also do a PR for the IHM02A1 to add a hint to the example

fpistm, 2019-02-25:
Why not.

closed 2019-02-26 by david-sawatzke