stm32duino/Arduino_Core_STM32 #458: Nucleo F401RE - Serial1 undefined

iacopo, 2019-02-26:
I'm very new to this and I know I should post to the stm32duino.com forum, but there's no way to register there!

I'm working on a nucleo F401RE, trying to use Serial1 in my sketch but the compiler says it's undefined.
I can only use Serial and Serial2, but they are basically the same one and mapped to the usb, I need an alternative serial communication port.

I thought there was some other error in my code, so I made a super simple one, but I confirm this does not work on my setup.

`long val = 0;

void setup() {
  // put your setup code here, to run once:
  //Serial.begin(115200); //works over USB
  //Serial2.begin(115200); //works over USB
  Serial1.begin(115200); //doesn't work
}

void loop() {
  // put your main code here, to run repeatedly:
  val++;
  Serial1.write(val); //doesn't work
  //Serial.write(val); //works over USB
  //Serial2.write(val); //works over USB
 
  delay(100);
}
`

here is a screenshot of my settings in the Arduino IDE

[image]

fpistm, 2019-02-26:
The generic instance on the Nucelo-F401RE is on `USART2`.
That's why `Serial2` is defined and mapped on `Serial`.

If you want another serial, you can add a build_opt.h file with: `-DENABLE_HWSERIAL1`
This will define the `Serial1 `using the first `USART1` instance found in the `PeripheralPins.c`.
In this case on `PA10`/`PA9`.
https://github.com/stm32duino/Arduino_Core_STM32/blob/master/variants/NUCLEO_F401RE/PeripheralPins.c#L153
https://github.com/stm32duino/Arduino_Core_STM32/blob/master/variants/NUCLEO_F401RE/PeripheralPins.c#L142

One other option is to simply define it:
`HardwareSerial Serial1(PA10, PA9);`
or 
`HardwareSerial Serial1(PB7, PB6);`
...

iacopo, 2019-02-26:
Thank you so much Frederic for the help, now I see my question was very trivial!

closed 2019-02-26 by iacopo

romainreignier, 2019-02-26:
I think this issue is natural for an Arduino user, I have faced it myself some weeks ago.
Maybe we should document it clearly somewhere.
Maybe here: https://github.com/stm32duino/wiki/wiki/API

iacopo, 2019-02-26:
Yes, documenting it would help a lot for new unexperienced users just like me

fpistm, 2019-02-27:
You're right guys, several documentation are missing.
I've planned to use the GitHub page feature to documents the core but time is always the missing ingredients. 😭 

Roger has created https://github.com/stm32duino/stm32duino.github.io 
This would be fine to use it. Then all will be available thanks stm32duino.github.io

romainreignier, 2019-02-27:
Ok nice, I can't promise but I might help a bit to kick start the doc.
So you would advise to use the plain website instead of the wiki?
Because I am not a web developer, I am Ok to write down some markdown, but not much to write plain HTML and CSS...
Or maybe we can find a nice documentation template for a static website generator like [jekyll](https://jekyllrb.com/) or [hugo](https://gohugo.io/hosting-and-deployment/hosting-on-github/).

fpistm, 2019-02-27:
Honestly, I don't know. Maybe a new issue can be opened to discuss about that and maybe some users could share their feedback about GitHub pages usage.