stm32duino/Arduino_Core_STM32 #460: Command sequence to obey breakpoints in GDB?

alexwhittemore, 2019-02-28:
> Any questions/feedback/suggestions should be discussed on the [stm32duino forum](http://stm32duino.com):
>   * questions on the [STM32 Core](http://stm32duino.com/viewforum.php?f=48)
>   * bugs/enhancements on the [STM core: Bugs and enhancements](http://stm32duino.com/viewforum.php?f=49)

^^ Unfortunately, this is no longer relevant. Sorry, here I am :)

I'm on Ubuntu 18.04, trying to use VSCode 1.31.1 with the latest core release 1.5.0. 

The general outline of my issue is that I'm trying to single-step an STM32F103 (Blue pill) program with an ST-Link.

I've gotten as far as getting the ARM toolchain set up, and the launch.json below will load code to flash and run it, but simply won't stop on breakpoints I set. I'm sure the command sequence in GDB is wrong, but I don't know enough about GDB to correct it. Hopefully it's something simple and someone knows it off hand.

My launch.json:

```
{
    "version": "0.2.0",
    "configurations": [
    
        {
            "name": "Arduino",
            "type": "arduino",
            "request": "launch",
            "program": "${file}",
            "cwd": "${workspaceRoot}",
            "MIMode": "gdb",
            "targetArchitecture": "arm",
            "miDebuggerPath": "/usr/bin/gdb-multiarch",
            "debugServerPath": "/usr/bin/openocd",
            "debugServerArgs": "-f interface/stlink-v2.cfg -f target/stm32f1x.cfg",
            "customLaunchSetupCommands": [
                {
                    "text": "set architecture arm"
                },
                {
                    "text": "target remote localhost:3333"
                },
                {
                    "text": "file ${file}"
                },
                {
                    "text": "load"
                },
                {
                    "text": "monitor reset halt"
                },
                {
                    "text": "monitor reset init"
                }
            ],
            "stopAtEntry": true,
            "serverStarted": "Info\\ :\\ [\\w\\d\\.]*:\\ hardware",
            "launchCompleteCommand": "exec-continue",
            "filterStderr": true,
            "args": [],
            "logging": { "engineLogging": true }
        }
    ]
}
```
I'm actually also trying to set this up on macOS, which uses arm-none-eabi-gdb, and where the "set architecture arm" command is unnecessary. So pardon me if I confuse platforms at some point - I'll try to avoid it.

If I read this correctly, That basically means

1) launch GDB
2) run commands in GDB:
```
>set architecture arm
>target remote localhost:3333
>file myapp.elf
>load
>monitor reset halt
>monitor reset init
```
It's the bits after "load" that I mostly don't understand. But the long and short of it is, the program just carries on running as if I have no breakpoints set.

If any of you is a (successful) vscode user, I'd sure appreciate your input!

BennehBoy, 2019-02-28:
If you're not massively tied to vscode there are instructions in the wiki
to get debugging working in eclipse, it works quite well

fpistm, 2019-02-28:
Hi @alexwhittemore, even if you could not register you could search on the forum ;)
http://stm32duino.com/viewtopic.php?f=41&t=2366&p=50363&hilit=vscode#p50363

Officially, we only support Arduino IDE. When I debug I mainly use directly gdb in command line.
Sometime I used Eclipse with Sloeber plugin as described here:
https://github.com/stm32duino/wiki/wiki/How-to-debug

I will let this issue open few days, if some user can help you but I've no clue with VSCode.

alexwhittemore, 2019-02-28:
@BennehBoy I definitely don't want to install another IDE, but that might have just enough info to get Code working anyway. Happy to borrow from other configurations - thanks for the pointer!

@fpistm Believe me, I've read all 6 pages. I'd post in that thread for clarity, but of course...

romainreignier, 2019-03-01:
Did you try cortex-debug plugin?
https://marketplace.visualstudio.com/items?itemName=marus25.cortex-debug
For me with ST CubeMX projects generated with Makefile it was working following the configurations found here:
https://marcelball.ca/projects/cortex-debug/cortex-debug-launch-configurations/

But I have never tried it with this Arduino core.

alexwhittemore, 2019-03-03:
Here you go - something nice and simple, that in retrospect makes perfect sense and not knowing what you're doing is a bug roadblock: In "Arduino Board configuration," "Optimize" needs to be set to "Debug (-g)" to add debug symbols to the output. Of course. With that and the launch.json above, debugging works a treat.

Oddly, nowhere in the 6 pages of that stm32duino.com thread is that mentioned. I finally figured it out stepping through the sloeber instructions at https://github.com/stm32duino/wiki/wiki/How-to-debug (thanks for leading me there, @BennehBoy - it held exactly the info I hoped for!)

Of course, in the process of getting this working, I've discovered like 6 different options for getting code on a blue pill and single stepping it. Luckily my preferred way is finally working, or I'd be paralyzed with choice!

Thanks for the help!

closed 2019-03-03 by alexwhittemore

alexwhittemore, 2019-03-03:
Ironic epilog: I continued following the sloeber instructions just for funsies after I'd gotten everything working in VSCode. It failed horribly - couldn't even compile without errors. Adding `#include <Arduino.h>` didn't fix it so I gave up and left it at that.

BennehBoy, 2019-03-03:
Yup sloeber doesn't like a macro in1.50:

```
\sloeber\arduinoPlugin\packages\STM32\hardware\stm32\1.5.0\cores\arduino/utils.h:9:26: note: in definition of macro 'CONCATS'
 #define CONCATS(s1, s2) (s1" " s2)
```

change to:

`#define CONCATS(s1, s2) (#s1 " " #s2)`

alexwhittemore, 2019-03-04:
Noted, thanks for mentioning it here!

fpistm, 2019-03-04:
What is strange is that you was not warned about this. When binary used to debug does not contains debug symbol, user is generally warned by gdb.
About issue with Sloeber, we already talk about it and unfortunately never see it. Moreover CONCAT is not new in 1.5.0 but this can be linked to which version of tools isused: Sloeber version, arm gcc, ...
@alexwhittemore does not described exactly which issue he met so difficult to answer what is wrong.
There are as many issues than user configs/env and unfortunately it's difficult to reproduce all. In a general way, user have to know/understand what/how/why. This is advanced usage.
I do not tell there is no bug :) Just that I saw several times issues or comments about this does not work without any description.