stm32duino/Arduino_Core_STM32 #461: STM32 support to ArduinoOTA library (help wanted)

jandrassy, 2019-02-28:
For Arduino SAMD and for nRF5 there is [ArduinoOTA library](https://github.com/jandrassy/ArduinoOTA), enhanced and maintained by me. I plan STM32 support, but I have too many Arduino plans. Someone could add STM32 support. The InternalStorage class stores the uploaded binary into upper half of the flash and then a function running in RAM copies the binary to standard location and resets the MCU. All know-how about the flash writing and reading is in the stm32duino EEPROM library.

PS: I would open this on forum, but there is no way to register and admins don't answer a message.

fpistm, 2019-02-28:
Hi @jandrassy 
I guess you should also open an issue on your side referring  this one.
Currently, I could not managed this soon, I'm very too busy.
Anyway if an other contributor(s) could hep, I will be happy to help/support him/them.

fpistm, 2020-01-03:
@jandrassy 
As there is an issue open to the Library, this should be tracked in this one.
Here this is mostly for issue with the core.
A topic about this can be open in the forum to discuss about it in the [STM32 CoreLibraries & Hardware](https://www.stm32duino.com/viewforum.php?f=39)

closed 2020-01-03 by fpistm

jandrassy, 2021-12-28:
I made support for F1xx