stm32duino/Arduino_Core_STM32 #463: With i2c new library doesn't work

charlie1024, 2019-02-28:
Hello,
I recently seen the board library was updated some days ago and the HAL feature was newly implemented.
I thought that with the new master branch board platform the new library will work, so with my Nucleo-F411RE I tried to operate the new library.

It worked without I2C. I use D14/D15(I2C1) and if I connect any I2C device, then the MCU goes to forever sleep(all codes are stopped). 

This code shows:
```
#include <Wire.h>
#include <STM32RTC.h>
STM32RTC& rtc = STM32RTC::getInstance();

void setup() {
  // put your setup code here, to run once:
  rtc.setClockSource(STM32RTC::LSE_CLOCK);
  rtc.begin(false); // initialize RTC 24H format

  if (!rtc.isTimeSet()) {
    rtc.setTime(0, 0, 0);
    rtc.setDate(1, 1, 2019); // fallback
  }
  pinMode(LED_BUILTIN, OUTPUT);

  Wire.begin();

  Wire.beginTransmission(0x50);
  Wire.write(0);
  Wire.endTransmission();
}

void loop() {
  // put your main code here, to run repeatedly:
  delay(500);
  digitalWrite(LED_BUILTIN, !digitalRead(LED_BUILTIN));
}
```

0x50 means the EEPROM device and I just wrote a waste data to the line, so there must be no problem.
However If I use this code with any I2C device connected, F411RE just halts and not respond for anything(interrupts are all stopped too).

Couldn't be the new library used yet? Or is this a bug?

Thank you.

fpistm, 2019-02-28:
@charlie1024,
I confirm this is a bug in the core since the last HAL update for F4.

fpistm, 2019-02-28:
As a workaround you can revert those 2 commits:
https://github.com/stm32duino/Arduino_Core_STM32/commit/df4966143edb3811c7f94fb906193417b9ccd4fb
https://github.com/stm32duino/Arduino_Core_STM32/commit/0e8722f4ffa6e18033f9b061b6c5bd93353e06de

charlie1024, 2019-03-01:
I tried to implement new RTC feature with [RTC] Avoid reset time after low power mode #442 and that did work in 1.5.0 platform. I hope the bug resolved. Thank you for assistance.

fpistm, 2019-03-01:
Of course this will be resolved ;)
Moreover there is a workaround to not block user who want use the master.

fpistm, 2019-03-05:
Issue found and fixed.

commit de686923 (2019-03-05): [I2C] Fix HAL F4

In function:
static HAL_StatusTypeDef I2C_Master_SB(I2C_HandleTypeDef *hi2c)

This check has been added:
      if ((hi2c->hdmatx->XferCpltCallback != NULL) || (hi2c->hdmarx->XferCpltCallback != NULL))
      {
        /* Enable DMA Request */
        SET_BIT(hi2c->Instance->CR2, I2C_CR2_DMAEN);
      }

But, in case of the DMA is not used, hi2c->hdmatx and hi2c->hdmarx are NULL.
In this case, hi2c->hdma(rx|tx)->XferCpltCallback can be not NULL
then I2C_CR2_DMAEN is set while it should not.
As this bit is checked several time to know if DMA is used or not this
raised an excepti [... cut]
closed 2019-03-05 by fpistm