stm32duino/Arduino_Core_STM32 #464: Add documentation for the particularities of this core

romainreignier, 2019-03-02:
To follow the discussion started in [#458](https://github.com/stm32duino/Arduino_Core_STM32/issues/458#issuecomment-467576466), it would be nice to document the usage of this core for users coming for official Arduino core.

Even if a dedicated website would be nice, it requires some work to setup a template with some style, categories... So instead, I propose to leverage the [existing wiki](https://github.com/stm32duino/wiki/wiki/).

I have started by adding the `HardwareSerial` info reported in #458, see here:
https://github.com/stm32duino/wiki/wiki/API#hardwareserial

Do you think it is the right place? I am open to any suggestions.

@fpistm For the `HardwareSerial` case, using the first method (declaring a `HardwareSerial` in the sketch) will not enable the `serialEvent()` callback, right?
For what I understand here:
https://github.com/stm32duino/Arduino_Core_STM32/blob/830bb00c7ee54fe8b7a0f7ca6edd5a8b44e5fbc7/cores/arduino/WSerial.h#L33-L37

Only passing `-DENABLE_HWSERIAL1` will enable it. Am I right?

fpistm, 2019-03-02:
Hi @romainreignier and thanks.
you're right for serialEvent.
Serial is the generic one map to the default serial instance defined int the variant if any.

romainreignier, 2019-03-02:
@fpistm Thank you, I have added a note.

So if you agree with the format, I will had the same kind of section for SPI and i2c.

fpistm, 2019-03-02:
Agreed of course 👍

romainreignier, 2019-03-04:
I have added i2c and SPI examples.

fpistm, 2019-05-03:
Note for me: upate upload method with #514

fpistm, 2019-07-23:
Since a month I've added/updated several Wiki pages.
I've added a ToDo list in the OP to track what is missing.

fpistm, 2020-01-31:
Wiki is updated regularly as requested by the OP.
I close this one as it is not used to track the Wiki which is updated in a particular repo.

closed 2020-01-31 by fpistm