stm32duino/Arduino_Core_STM32 #465: Warning: changing start of section .bss by x bytes

lorol, 2019-03-05:
**Info**
 - OS: Win 10
 - Arduino IDE version: 1.8.8
 - STM32 core version: 1.4.0 and 1.5.0
 - Upload method: Mass Storage

**Board**
 - Name: Nucleo L152RE

**Additional context**
Here is an .ino file (it needs few lib's but you can figure out) to reproduce the warning.
https://gist.github.com/b41c43db7f3ac0990b5639921658f024.git
The cause is a double (64-bit) var at line 96. If you are using float, no problems.
If you move the position where it is declared, it may affect the warning output (or not)
It works ... but the difference is: with warning it takes +4 RAM bytes more :)
With corrected .bss ALIGN(4) no warnings and less memory.

Related commit: 
https://github.com/stm32duino/Arduino_Core_STM32/commit/d52e577c5672af925c66d25112af9667eb5eb745#commitcomment-32583137

fpistm, 2019-03-27:
@lorol 
I have not exactly the same result with your example.
With no `.bss` alignment I have:
```
Sketch uses 53696 bytes (10%) of program storage space. Maximum is 524288 bytes.
Global variables use 4596 bytes (5%) of dynamic memory, leaving 77324 bytes for local variables. Maximum is 81920 bytes.
```
while with `ALIGN(4)`:
```
Sketch uses 53696 bytes (10%) of program storage space. Maximum is 524288 bytes.
Global variables use 4600 bytes (5%) of dynamic memory, leaving 77320 bytes for local variables. Maximum is 81920 bytes.
```
So I lost 4 bytes with alignment.

What I will do is to remove the warn section alignment as by default, SECTIONS command
does not specify a start address for the section, let ld do the work.
If a start address is specified no warn was displayed.

lorol, 2019-03-27:
Great!
Makes sense, if you solved the warning and the RAM is actually better managed by ld  - no problems.
However, on my builds:

IDE 1.8.8 STM32 1.5.0 NUCLEO L152
one double variable in sketch

-----------------ORIGINAL ld---------------------------
  /* Uninitialized data section */
  . = ALIGN(4);
  **.bss :**

./../../arm-none-eabi/bin/ld.exe: warning: changing start of section .bss by 4 bytes

Sketch uses 54552 bytes (10%) of program storage space. Maximum is 524288 bytes.
Global variables use **5016** bytes (6%) of dynamic memory, leaving 76904 bytes for local variables. Maximum is 81920 bytes.

----------------PATCHED ld----------------------------
  /* Uninitialized data section */
  . = ALIGN(4);
  **.bss ALIGN(4) :**

Sketch uses 54552 bytes (10%) of program storage space. Maximum is 524288 bytes.
Global variables use **5012** bytes (6%) of dynamic memory, leaving 76908 bytes for local variables. Maximum is 81920 bytes.

closed 2019-03-27 by fpistm

lorol, 2019-03-27:
Great!

PR #484 (2019-03-27): Do not warn if start of section is changed
Remove `ld `option '`warn-section-align`', as by default, **SECTIONS** command
does not specify a start address for the section, let `ld `do the work.
If a start address is specified no warn was displayed.

> --warn-section-align
>     Warn if the address of an output section is changed because of
>     alignment. Typically, the alignment will be set by an input section.
>     The address will only be changed if it not explicitly specified;
>     that is, if the SECTIONS command does not specify a start address
>     for the section.

Fix #465
commit 9299ab27 (2019-03-27): Do not warn if start of section is changed

Remove ld option 'warn-section-align', as by default, SECTIONS command
does not specify a start address for the section, let ld do the work.
If a start address is specified no warn was displayed.

--warn-section-align
    Warn if the address of an output section is changed because of
    alignment. Typically, the alignment will be set by an input section.
    The address will only be changed if it not explicitly specified;
    that is, if the SECTIONS command does not specify a start address
    for the section.

Fix #465