stm32duino/Arduino_Core_STM32 #471: Should links/forwarding to the old read-only forum be disabled?

hierophect, 2019-03-12:
I am a new user of the STM32duino on a Bluepill board. I have been finding it frustrating how many of the official support materials here suggest or outright require starting discussions on a forum that does not allow new user registration and will soon be made read-only (April). I suppose I'm a bit extra grumpy as I know that many older users are still able to use this forum for this purpose, just not new ones like me ;)

I would suggest at the very least, the links in the Troubleshooting section of the readme should be changed to make it clear that all issues and discussions are posted on the github, rather than referring to a defunct forum. The contributing guidelines should also be changed to reflect that there is no discussion possible for new users on the old forum and no questions can be asked there. If there is in fact a new forum being used, then they should link to that, though I have not seen one on the ST site or elsewhere. 

If the forum is in fact being archived, is it expected that a new forum will open up elsewhere? Or will the contributing guidelines for this github be significantly loosened to allow "casual" conversation? I have some questions I would like to ask about the compatibility of USB serial communication across this and Roger's old version, which I would not normally consider a github appropriate issue. If that's not appropriate, is there any plan in place for a new general discussion hub once the old forum is closed permanently?

hierophect, 2019-03-12:
I would recommend, for instance, having all materials link to this new forum which has apparently been started by a community member so that questions can actually be answered: https://mcu.selfip.com/viewforum.php?f=5

fpistm, 2019-03-13:
Hi @hierophect 
Actually, this is one of my main concerns to find the best way to provide support.
And working hard on this subject.
I would/could not redirect to any new forum while all alternative solutions reviewed and all people agreed.
In the case you proposed @Serasidis is the owner and maybe would not like STM32duino GitHub organization redirect to his forum.

BennehBoy, 2019-03-13:
@fpistm not sure if you have the bandwidth, but new users are beginning to ask STM core related questions there, may be worth registering 😄

hierophect, 2019-03-13:
Am I correct in assuming this is the official, ST backed stm32duino core, being discussed as the "future" on the old forum? It might also be a decent idea to document just a little bit of the history of the project in the ReadMe so people understand what the situation is. It isn't listed as a fork of Roger's code so it's a bit confusing. I'm still not totally clear on how the two projects are related.

BennehBoy, 2019-03-13:
It's not a fork of Roger's code.

The only real relationship between the 2 projects is they are for STM32 devices using the Arduino IDE.

There's some crossover between them due to some of the same folks using both.

Roger recently gifted the stm32duino.com domain to STM since this 'official' core has 'official' support, ie @fpistm who is an STM employee.  The rest of us are just interested parties.

A history page on the Wiki might be useful, but others other than I are probably better qualified/more in the know to come up with it.

hierophect, 2019-03-13:
Thanks for the explanation Benneh. 

I've taken the time to read the threads on the old forum discussing the closure/move, and it seems as though the conversation is circling around trying to pick a name for a new community and forward everyone to it. However, I missed that the actual stm32duino.com domain (as opposed to just the github) had been transferred. @fpistm, is it probable that ST will be backing a new community based on this domain name, similar to AVRFreaks? 

While I'm a newcomer to the platform, I've been very pleased at how quickly I've been able to get features like debugging and low-power features working on my ST chips with this core (especially when compared to mbed, the other common "upgrade path" off arduino), and I'd like to recommend it as I release projects based on it. However, I'm very uncomfortable promoting any framework that lacks a "landing area" for newcomers. 

I'd hope we could either get a decisive green light from ST that they'll back a new forum with the stm32duino domain name, or an equivalently decisive no, so that a new community backed forum can be established with confidence.

BennehBoy, 2019-03-13:
Whilst I understand the frustrations for people right now, I don't think there's too much reason to worry, there's already the above forum where all are welcome (otherwise Vassilis would not have announced it and created it on the public stm32duino.com forum), and if that turns out to be a stopgap it's no big deal.

hierophect, 2019-03-13:
I don't mean to sound too annoyed, it just took me a while to discover all these details, and it mostly seems that I hopped on this framework at exactly the wrong time. 

However, I'd still advocate for the official Readme and contributing guidelines being changed for this interim period. Even if a resolution is expected within the next few months, if a new user lands on this github, they shouldn't have to dig into a months worth of threads to find out why they can't post questions.

commit ab2e9869 (2019-03-14): Update README.md

Fix #471
closed 2019-03-14 by fpistm