stm32duino/Arduino_Core_STM32 #476: How to use Serial2

nopnop2002, 2019-03-22:
I found this section on your boards.txt.

	# NUCLEO_F103RB board
	# Support: Serial1 (USART1 on PA10, PA9) and Serial2 (USART3 on PC11, PC10)

I tried Serial2.
But Serial2.printf() output to Arduino-IDE Monitor.
I want to output link this.
To Arduino-IDE using Serial.
To PC10 using Serial2.

Code:

	void setup() {

	  Serial.begin(115200);
	  Serial2.begin(115200);
	  Serial.println("setup() start");
	  Serial2.println("Serial2 setup() start");

	  while (1) {
	    Serial.println("setup alive...");
	    Serial2.println("Serial2 setup alive...");
	    delay(1000);
	  }
	}

	void loop() {
	  // Not used.
	}

fpistm, 2019-03-22:
Hum in fact, this is not right, copy/paste mistake.
By default Serial is mapped to Serial2 on PA3/PA2.
https://github.com/stm32duino/Arduino_Core_STM32/blob/c153f5fb929603926adbafb7a0b5c87b07a3a551/variants/NUCLEO_F103RB/variant.h#L114-L119

Other U(S)ART possibility are here:
https://github.com/stm32duino/Arduino_Core_STM32/blob/c153f5fb929603926adbafb7a0b5c87b07a3a551/variants/NUCLEO_F103RB/PeripheralPins.c#L158-L178

Some info:
https://github.com/stm32duino/wiki/wiki/API#hardwareserial

closed 2019-03-22 by fpistm

nopnop2002, 2019-03-22:
Serial1 seems to work right.
I confirm that Serial1.println() output to PA9.

My code:

	HardwareSerial Serial1(PA10, PA9);

	void setup() {

	  Serial.begin(115200);
	  Serial1.begin(115200);
	  Serial.println("setup() start");
	  Serial1.println("Serial1 setup() start");

	  while (1) {
	    Serial.println("setup alive...");
	    Serial1.println("Serial1 setup alive...");
	    delay(1000);
	  }
	}

	void loop() {
	  // Not used.
	}

Thank you.

commit 7feeb092 (2019-03-22): Remove boards.txt comments to avoid misalignment

Board variant files describe the board support.
Fix #476