stm32duino/Arduino_Core_STM32 #479: Discovery L072CZ USB Serial

evandavey, 2019-03-25:
**Describe the bug**
I get an Unknown USB Device (Device Descriptor Request Failed) when trying to use CDC Serial on a Discovery L072CZ-LWRAN1.  I know the USB works as I've successfully used it with this core https://github.com/GrumpyOldPizza/ArduinoCore-stm32l0.

**To Reproduce**
Load blink example, selecting USB CDC support from the submenu.

fpistm, 2019-03-25:
Hi @evandavey 
Right, this do not work as the USB CLK is not enabled by default in the `SystemClock_Config()`

You can add this to your sketch as it is a weak function, this will enable the USB clock:

```
extern "C" void SystemClock_Config(void)
{
  RCC_OscInitTypeDef RCC_OscInitStruct;
  RCC_ClkInitTypeDef RCC_ClkInitStruct;
  RCC_PeriphCLKInitTypeDef PeriphClkInit;

  __HAL_RCC_PWR_CLK_ENABLE();

  __HAL_PWR_VOLTAGESCALING_CONFIG(PWR_REGULATOR_VOLTAGE_SCALE1);

  RCC_OscInitStruct.OscillatorType = RCC_OSCILLATORTYPE_HSI|RCC_OSCILLATORTYPE_HSI48;
  RCC_OscInitStruct.HSIState = RCC_HSI_ON;
  RCC_OscInitStruct.HSICalibrationValue = RCC_HSICALIBRATION_DEFAULT;
  RCC_OscInitStruct.HSI48State = RCC_HSI48_ON;
  RCC_OscInitStruct.PLL.PLLState = RCC_PLL_ON;
  RCC_OscInitStruct.PLL.PLLSource = RCC_PLLSOURCE_HSI;
  RCC_OscInitStruct.PLL.PLLMUL = RCC_PLLMUL_4;
  RCC_OscInitStruct.PLL.PLLDIV = RCC_PLLDIV_2;
  if (HAL_RCC_OscConfig(&RCC_OscInitStruct) != HAL_OK) {
    Error_Handler();
  }

  RCC_ClkInitStruct.ClockType = RCC_CLOCKTYPE_HCLK | RCC_CLOCKTYPE_SYSCLK
                                | RCC_CLOCKTYPE_PCLK1 | RCC_CLOCKTYPE_PCLK2;
  RCC_ClkInitStruct.SYSCLKSource = RCC_SYSCLKSOURCE_PLLCLK;
  RCC_ClkInitStruct.AHBCLKDivider = RCC_SYSCLK_DIV1;
  RCC_ClkInitStruct.APB1CLKDivider = RCC_HCLK_DIV1;
  RCC_ClkInitStruct.APB2CLKDivider = RCC_HCLK_DIV1;
  if (HAL_RCC_ClockConfig(&RCC_ClkInitStruct, FLASH_LATENCY_1) != HAL_OK) {
    Error_Handler();
  }
  PeriphClkInit.PeriphClockSelection = RCC_PERIPHCLK_USB;
  PeriphClkInit.UsbClockSelection = RCC_USBCLKSOURCE_HSI48;
  if (HAL_RCCEx_PeriphCLKConfig(&PeriphClkInit) != HAL_OK) {
    Error_Handler();
  }

  HAL_SYSTICK_Config(HAL_RCC_GetHCLKFreq() / 1000);

  HAL_SYSTICK_CLKSourceConfig(SYSTICK_CLKSOURCE_HCLK);

  /* SysTick_IRQn interrupt configuration */
  HAL_NVIC_SetPriority(SysTick_IRQn, 0, 0);
}

```

evandavey, 2019-03-25:
OK, thanks.  I will give this a go tomorrow and close the thread if successful.

fpistm, 2019-03-25:
@evandavey I've tested and it works. I've also push a PR to enable USB clock by default even if by default USB could not work as SB15/16 are not fitted (Don't ask me why, I don't know 🤣 )

evandavey, 2019-03-25:
OK great!  I am actually on a custom board using the CMWX1ZZABZ-078 module and using this board variant as a starting point.  Do you know if arduino-lmic works?

evandavey, 2019-03-25:
So this works if I use a simple example like blink.  However, I tried adding it to the LMIC TTN-OTAA example and it compiles and uploads but I then get the same device not recognized error.

fpistm, 2019-03-26:
@evandavey 
which library you used ?
And could you be more precise about the board you really used ?

evandavey, 2019-03-26:
Board is a custom board we have designed using the Murata CMWX1ZZABZ-78
module. Set to use internal clock. Library was arduino-lmic latest version
installed via library manager (away from pc so can't check exact).

fpistm, 2019-03-26:
https://github.com/mcci-catena/arduino-lmic?

fpistm, 2019-03-26:
Well, reproduce your issue with this lib.
Adding `while(!Serial);` allow to have a proper enumeration.
In fact,     `os_init();` seems to disable irq before USB ended to configure.
Anyway, I think some update will be required to be used with this core.

closed 2019-03-26 by fpistm

evandavey, 2019-03-26:
OK, this is what I was using https://github.com/matthijskooijman/arduino-lmic.git.  Will try some other libraries and see if anything works.  In the meantime, I will probably switch over to MBED.

fpistm, 2019-03-26:
No worries. This issue was closed automatically after merge of the #481, which fix original issue.
Arduino-lmic is an other request.

PR #481 (2019-03-26): [DISCO_L072CZ_LRWAN1] Enable USB clock
Fix #479

As the board has user USB connector, enable it by default.
Anyway, SB25 and SB16 have to be closed to have USB functional.
commit 59673c97 (2019-03-26): [DISCO_L072CZ_LRWAN1] Enable USB clock

Fix #479