stm32duino/Arduino_Core_STM32 #482: HardwareSerial data bits issue?

mrv96, 2019-03-25:
Hi, i'm analyzing HardwareSerial library, inside begin method i've seen that the variable databits can assume from config the values 6,7,8, but HAL library supports only 7,8,9 data bits. Should the defined configurations be changed and subsequently change the if inside begin method?

fpistm, 2019-03-25:
IIWR, 6 databits could be managed with a parity bit on some STM32.
Do not mix uart data and uart frame.

Ex for F3, for some device you can have 6 bits + PB:

https://github.com/stm32duino/Arduino_Core_STM32/blob/7feeb092bcaeed4d01f6de0e35b1666b2c4cefd3/system/Drivers/STM32F3xx_HAL_Driver/Inc/stm32f3xx_hal_usart_ex.h#L64-L67

https://github.com/stm32duino/Arduino_Core_STM32/blob/7feeb092bcaeed4d01f6de0e35b1666b2c4cefd3/system/Drivers/STM32F3xx_HAL_Driver/Src/stm32f3xx_hal_uart_ex.c#L115-L147

mrv96, 2019-03-25:
Ok sorry, my fault. In any case, for a generic user, it seems that 9 data bits are not achievable by the library, because both defined config miss and databits cannot be 9.

fpistm, 2019-03-25:
Right, 9 is not achievable.
As Arduino doesn't define it it was not added
https://www.arduino.cc/reference/en/language/functions/communication/serial/begin/
Moreover, HardwareSerial::write allows only uint8_t.

mrv96, 2019-03-25:
Ok... i believe that it would better implement it because STM32 is a very powerfull uC. If the software philosopy make it equal as Arduino, why should i use STM32? :) Changing slightly the library, in my current application i use 9 data bits, however it would be fantastic if also less expert people can use the whole STM32 power :)

fpistm, 2019-03-25:
First goal of this core was to be able to use STM32 based boards with Arduino and to be as more as possible compatible with basics sketches and libraries.
I think this step is achieved 🙄, now core features are/could be extended and also performance/robustness.
All HAL and LL are available at  level sketch some advanced user already use it to use further capabilities.

Adding the 9bits support could be a good addition. Do not hesitate to submit a PR.
Any contribution are welcome :)

fpistm, 2019-04-02:
Close it as no issue.

closed 2019-04-02 by fpistm