stm32duino/Arduino_Core_STM32 #483: Unable to compile build_opt.h invalid argument

zhenninglim, 2019-03-26:
arm-none-eabi-g++:error:@C:\build/sketch/build_opt.h: invalid argument. 

STM32 1.5.0
STM32 tools 1.2.1

I am getting this problem when i tried to compile in windows 10.

fpistm, 2019-03-26:
Hi @zhenninglim ,
Could you describe a little more?
Which Arduino IDE version you use? Which sketch? Which board?

It seems you have an "@" in the path and I could not understand why at this time without more description.

zhenninglim, 2019-03-26:
Hi fpistm, 
I am using 

arduino 1.5.8. 
Board generic stm32f0 series
Optimize smallest - 0s
Board no stm32f030f4 demo board
Runtime newlib nano

I tried to compile the blink example.

fpistm, 2019-03-26:
1.5.8? or 1.8.5?
Could you enable in the `Preferences` the  `compilation` for `Show verbose output during` and also all `Compiler warnings`. Then copy/paste the output log.
Do you add a `build_opt.h` file to the Blink sketch ?

zhenninglim, 2019-03-26:
Version 1.5.8.  Attached is the log

```
Build options changed, rebuilding all

C:\Users\-\Desktop\IDE1.5.8C\IDE1.5.8C/hardware/tools/gcc-arm-none-eabi-6-2017-q2-update/bin/arm-none-eabi-g++ -mcpu=cortex-m0 -mthumb @C:\Users\-\Documents\build/sketch/build_opt.h -c -Os -w -std=gnu++14 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -MMD -I{build.source.path} -IC:\Users\-\Desktop\IDE1.5.8C\IDE1.5.8C\hardware\STM32\sam\cores\arduino/avr -IC:\Users\-\Desktop\IDE1.5.8C\IDE1.5.8C\hardware\STM32\sam\cores\arduino/stm32 -IC:\Users\-\Desktop\IDE1.5.8C\IDE1.5.8C\hardware\STM32\sam\cores\arduino/stm32/LL -IC:\Users\-\Desktop\IDE1.5.8C\IDE1.5.8C\hardware\STM32\sam\cores\arduino/stm32/usb -IC:\Users\-\Desktop\IDE1.5.8C\IDE1.5.8C\hardware\STM32\sam\cores\arduino/stm32/usb/hid -IC:\Users\-\Desktop\IDE1.5.8C\IDE1.5.8C\hardware\STM32\sam\cores\arduino/stm32/usb/cdc -IC:\Users\-\Desktop\IDE1.5.8C\IDE1.5.8C\hardware\STM32\sam\system/Drivers/CMSIS/Include/ -IC:\Users\-\Desktop\IDE1.5.8C\IDE1.5.8C\hardware\STM32\sam\system/Drivers/STM32F0xx_HAL_Driver/Inc/ -IC:\Users\-\Desktop\IDE1.5.8C\IDE1.5.8C\hardware\STM32\sam\system/Drivers/STM32F0xx_HAL_Driver/Src/ -IC:\Users\-\Desktop\IDE1.5.8C\IDE1.5.8C\hardware\STM32\sam\system/STM32F0xx/ -IC:\Users\-\Desktop\IDE1.5.8C\IDE1.5.8C\hardware\STM32\sam\system/Middlewares/ST/STM32_USB_Device_Library/Core/Inc -IC:\Users\-\Desktop\IDE1.5.8C\IDE1.5.8C\hardware\STM32\sam\system/Middlewares/ST/STM32_USB_Device_Library/Core/Src -DSTM32F0xx -DARDUINO=158 -DARDUINO_DEMO_F030F4 -DARDUINO_ARCH_SAM -DBOARD_NAME="DEMO_F030F4" -DSTM32F030x6 -DHAL_UART_MODULE_ENABLED -I{runtime.tools.CMSIS-5.3.0.path}/CMSIS/Core/Include/ -IC:\Users\-\Desktop\IDE1.5.8C\IDE1.5.8C\hardware\STM32\sam\system/Drivers/CMSIS/Device/ST/STM32F0xx/Include/ -IC:\Users\-\Desktop\IDE1.5.8C\IDE1.5.8C\hardware\STM32\sam\system/Drivers/CMSIS/Device/ST/STM32F0xx/Source/Templates/gcc/ -IC:\Users\-\Desktop\IDE1.5.8C\IDE1.5.8C\hardware\STM32\sam\cores\arduino -IC:\Users\-\Desktop\IDE1.5.8C\IDE1.5.8C\hardware\STM32\sam\variants\DEMO_F030F4 C:\Users\-\Documents\build\st.cpp -o C:\Users\-\Documents\build\st.cpp.o 
arm-none-eabi-g++: error: @C:\Users\-\Documents\build/sketch/build_opt.h: Invalid argument
Error compiling.
```

fpistm, 2019-03-26:
Do you have a space in your user name ?

zhenninglim, 2019-03-26:
Nope

fpistm, 2019-03-26:
Ok so please try with the latest Arduino IDE. 
1.5.8 is too old (2014) to manage properly the platform.txt (pre build and other recipe).
We do not support this version.

zhenninglim, 2019-03-26:
Ok.  I shall try with 1.8. Will get back on this.

zhenninglim, 2019-03-26:
Just tried it out.  Manage to resolve the build_opt.h issue.  The Arduino 1.5.8 could not manage the platform.txt properly.  Currently,  there is another issue cannot find -larm_cortexM01_math.  Thanks for helping out

fpistm, 2019-03-26:
I guess you would mean `arm_cortexM0l_math` not `arm_cortexM0l_math`
This is provided by the CMSIS package installed with the core when using the board manager.
Without more information from you I can not help.
I guess this is due how you install the core in your setup.

zhenninglim, 2019-03-26:
Got it.  Thanks!

closed 2019-03-26 by zhenninglim