stm32duino/Arduino_Core_STM32 #486: F103 USB CDC won't build

cdrose, 2019-03-28:
STM32F103C8T6 BluePill
Eclipse with Sloeber plugin
Arduino Core STM32 1.5.0 latest
Windows 10

I'm getting a string of errors if I try to include USB CDC into my project. It looks somewhat like there's a missing #define and then a bunch of other macros don't expand properly, but I don't know enough about how this is all meant to work to spot the problem. I see from other comments people are using CDC so it must have been working before, or my build environment isn't set up correctly.

Any ideas?

`"C:\\eclipse\\arduinoPlugin\\tools\\make\\make" -j4 all 
C:/eclipse/arduinoPlugin/tools/make/make --no-print-directory pre-build
cmd /c if not exist "/build_opt.h" mkdir "C:\eclipse-workspace\Blink/Release\sketch" & type NUL > "C:\eclipse-workspace\Blink/Release/sketch/build_opt.h"
A subdirectory or file C:\eclipse-workspace\Blink/Release\sketch already exists.
' '
C:/eclipse/arduinoPlugin/tools/make/make --no-print-directory main-build
'Building file: C:\eclipse\arduinoPlugin\packages\STM32\hardware\stm32\1.5.0\cores\arduino\stm32\usb\usbd_desc.c'
'Building file: C:\eclipse\arduinoPlugin\packages\STM32\hardware\stm32\1.5.0\cores\arduino\stm32\LL\stm32yyxx_ll_bdma.c'
'Starting C compile'
'Starting C compile'
"C:\eclipse\arduinoPlugin\packages\STM32\tools\arm-none-eabi-gcc\6-2017-q2-update/bin/arm-none-eabi-gcc" -mcpu=cortex-m3 -mthumb "@C:/eclipse-workspace/Blink/Release/sketch/build_opt.h" -c -g -Og -Wall -Wextra -std=gnu11 -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -MMD "-I" "-IC:\eclipse\arduinoPlugin\packages\STM32\hardware\stm32\1.5.0\cores\arduino/avr" "-IC:\eclipse\arduinoPlugin\packages\STM32\hardware\stm32\1.5.0\cores\arduino/stm32" "-IC:\eclipse\arduinoPlugin\packages\STM32\hardware\stm32\1.5.0\cores\arduino/stm32/LL" "-IC:\eclipse\arduinoPlugin\packages\STM32\hardware\stm32\1.5.0\cores\arduino/stm32/usb" "-IC:\eclipse\arduinoPlugin\packages\STM32\hardware\stm32\1.5.0\cores\arduino/stm32/usb/hid" "-IC:\eclipse\arduinoPlugin\packages\STM32\hardware\stm32\1.5.0\cores\arduino/stm32/usb/cdc" "-IC:\eclipse\/arduinoPlugin/packages/STM32/hardware/stm32/1.5.0/system/Drivers/STM32F1xx_HAL_Driver/Inc/" "-IC:\eclipse\/arduinoPlugin/packages/STM32/hardware/stm32/1.5.0/system/Drivers/STM32F1xx_HAL_Driver/Src/" "-IC:\eclipse\/arduinoPlugin/packages/STM32/hardware/stm32/1.5.0/system/STM32F1xx/" "-IC:\eclipse\/arduinoPlugin/packages/STM32/hardware/stm32/1.5.0/system/Middlewares/ST/STM32_USB_Device_Library/Core/Inc" "-IC:\eclipse\/arduinoPlugin/packages/STM32/hardware/stm32/1.5.0/system/Middlewares/ST/STM32_USB_Device_Library/Core/Src" -DSTM32F1xx -DARDUINO=10802 -DARDUINO_BLUEPILL_F103C8 -DARDUINO_ARCH_STM32 -DBOARD_NAME="BLUEPILL_F103C8"  -DSTM32F103xB -DUSBCON  -DUSBD_VID=0x0483 "-DUSB_MANUFACTURER=\"Unknown\"" "-DUSB_PRODUCT="BLUEPILL_F103C8"" -DHAL_PCD_MODULE_ENABLED -DUSBD_USE_CDC  "-IC:\eclipse\arduinoPlugin\packages\STM32\tools\CMSIS\5.3.0/CMSIS/Core/Include/" "-IC:\eclipse\/arduinoPlugin/packages/STM32/hardware/stm32/1.5.0/system/Drivers/CMSIS/Device/ST/STM32F1xx/Include/" "-IC:\eclipse\/arduinoPlugin/packages/STM32/hardware/stm32/1.5.0/system/Drivers/CMSIS/Device/ST/STM32F1xx/Source/Templates/gcc/"   -I"C:\eclipse\arduinoPlugin\packages\STM32\hardware\stm32\1.5.0\cores\arduino" -I"C:\eclipse\arduinoPlugin\packages\STM32\hardware\stm32\1.5.0\variants\BLUEPILL_F103XX" -MMD -MP -MF"core\core\stm32\usb\usbd_desc.c.d" -MT"core\core\stm32\usb\usbd_desc.c.o" -D__IN_ECLIPSE__=1 "C:\eclipse\arduinoPlugin\packages\STM32\hardware\stm32\1.5.0\cores\arduino\stm32\usb\usbd_desc.c"  -o  "core\core\stm32\usb\usbd_desc.c.o"
'Building file: C:\eclipse\arduinoPlugin\packages\STM32\hardware\stm32\1.5.0\cores\arduino\stm32\LL\stm32yyxx_ll_comp.c'
"C:\eclipse\arduinoPlugin\packages\STM32\tools\arm-none-eabi-gcc\6-2017-q2-update/bin/arm-none-eabi-gcc" -mcpu=cortex-m3 -mthumb "@C:/eclipse-workspace/Blink/Release/sketch/build_opt.h" -c -g -Og -Wall -Wextra -std=gnu11 -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -MMD "-I" "-IC:\eclipse\arduinoPlugin\packages\STM32\hardware\stm32\1.5.0\cores\arduino/avr" "-IC:\eclipse\arduinoPlugin\packages\STM32\hardware\stm32\1.5.0\cores\arduino/stm32" "-IC:\eclipse\arduinoPlugin\packages\STM32\hardware\stm32\1.5.0\cores\arduino/stm32/LL" "-IC:\eclipse\arduinoPlugin\packages\STM32\hardware\stm32\1.5.0\cores\arduino/stm32/usb" "-IC:\eclipse\arduinoPlugin\packages\STM32\hardware\stm32\1.5.0\cores\arduino/stm32/usb/hid" "-IC:\eclipse\arduinoPlugin\packages\STM32\hardware\stm32\1.5.0\cores\arduino/stm32/usb/cdc" "-IC:\eclipse\/arduinoPlugin/packages/STM32/hardware/stm32/1.5.0/system/Drivers/STM32F1xx_HAL_Driver/Inc/" "-IC:\eclipse\/arduinoPlugin/packages/STM32/hardware/stm32/1.5.0/system/Drivers/STM32F1xx_HAL_Driver/Src/" "-IC:\eclipse\/arduinoPlugin/packages/STM32/hardware/stm32/1.5.0/system/STM32F1xx/" "-IC:\eclipse\/arduinoPlugin/packages/STM32/hardware/stm32/1.5.0/system/Middlewares/ST/STM32_USB_Device_Library/Core/Inc" "-IC:\eclipse\/arduinoPlugin/packages/STM32/hardware/stm32/1.5.0/system/Middlewares/ST/STM32_USB_Device_Library/Core/Src" -DSTM32F1xx -DARDUINO=10802 -DARDUINO_BLUEPILL_F103C8 -DARDUINO_ARCH_STM32 -DBOARD_NAME="BLUEPILL_F103C8"  -DSTM32F103xB -DUSBCON  -DUSBD_VID=0x0483 "-DUSB_MANUFACTURER=\"Unknown\"" "-DUSB_PRODUCT="BLUEPILL_F103C8"" -DHAL_PCD_MODULE_ENABLED -DUSBD_USE_CDC  "-IC:\eclipse\arduinoPlugin\packages\STM32\tools\CMSIS\5.3.0/CMSIS/Core/Include/" "-IC:\eclipse\/arduinoPlugin/packages/STM32/hardware/stm32/1.5.0/system/Drivers/CMSIS/Device/ST/STM32F1xx/Include/" "-IC:\eclipse\/arduinoPlugin/packages/STM32/hardware/stm32/1.5.0/system/Drivers/CMSIS/Device/ST/STM32F1xx/Source/Templates/gcc/"   -I"C:\eclipse\arduinoPlugin\packages\STM32\hardware\stm32\1.5.0\cores\arduino" -I"C:\eclipse\arduinoPlugin\packages\STM32\hardware\stm32\1.5.0\variants\BLUEPILL_F103XX" -MMD -MP -MF"core\core\stm32\LL\stm32yyxx_ll_bdma.c.d" -MT"core\core\stm32\LL\stm32yyxx_ll_bdma.c.o" -D__IN_ECLIPSE__=1 "C:\eclipse\arduinoPlugin\packages\STM32\hardware\stm32\1.5.0\cores\arduino\stm32\LL\stm32yyxx_ll_bdma.c"  -o  "core\core\stm32\LL\stm32yyxx_ll_bdma.c.o"
'Starting C compile'
"C:\eclipse\arduinoPlugin\packages\STM32\tools\arm-none-eabi-gcc\6-2017-q2-update/bin/arm-none-eabi-gcc" -mcpu=cortex-m3 -mthumb "@C:/eclipse-workspace/Blink/Release/sketch/build_opt.h" -c -g -Og -Wall -Wextra -std=gnu11 -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -MMD "-I" "-IC:\eclipse\arduinoPlugin\packages\STM32\hardware\stm32\1.5.0\cores\arduino/avr" "-IC:\eclipse\arduinoPlugin\packages\STM32\hardware\stm32\1.5.0\cores\arduino/stm32" "-IC:\eclipse\arduinoPlugin\packages\STM32\hardware\stm32\1.5.0\cores\arduino/stm32/LL" "-IC:\eclipse\arduinoPlugin\packages\STM32\hardware\stm32\1.5.0\cores\arduino/stm32/usb" "-IC:\eclipse\arduinoPlugin\packages\STM32\hardware\stm32\1.5.0\cores\arduino/stm32/usb/hid" "-IC:\eclipse\arduinoPlugin\packages\STM32\hardware\stm32\1.5.0\cores\arduino/stm32/usb/cdc" "-IC:\eclipse\/arduinoPlugin/packages/STM32/hardware/stm32/1.5.0/system/Drivers/STM32F1xx_HAL_Driver/Inc/" "-IC:\eclipse\/arduinoPlugin/packages/STM32/hardware/stm32/1.5.0/system/Drivers/STM32F1xx_HAL_Driver/Src/" "-IC:\eclipse\/arduinoPlugin/packages/STM32/hardware/stm32/1.5.0/system/STM32F1xx/" "-IC:\eclipse\/arduinoPlugin/packages/STM32/hardware/stm32/1.5.0/system/Middlewares/ST/STM32_USB_Device_Library/Core/Inc" "-IC:\eclipse\/arduinoPlugin/packages/STM32/hardware/stm32/1.5.0/system/Middlewares/ST/STM32_USB_Device_Library/Core/Src" -DSTM32F1xx -DARDUINO=10802 -DARDUINO_BLUEPILL_F103C8 -DARDUINO_ARCH_STM32 -DBOARD_NAME="BLUEPILL_F103C8"  -DSTM32F103xB -DUSBCON  -DUSBD_VID=0x0483 "-DUSB_MANUFACTURER=\"Unknown\"" "-DUSB_PRODUCT="BLUEPILL_F103C8"" -DHAL_PCD_MODULE_ENABLED -DUSBD_USE_CDC  "-IC:\eclipse\arduinoPlugin\packages\STM32\tools\CMSIS\5.3.0/CMSIS/Core/Include/" "-IC:\eclipse\/arduinoPlugin/packages/STM32/hardware/stm32/1.5.0/system/Drivers/CMSIS/Device/ST/STM32F1xx/Include/" "-IC:\eclipse\/arduinoPlugin/packages/STM32/hardware/stm32/1.5.0/system/Drivers/CMSIS/Device/ST/STM32F1xx/Source/Templates/gcc/"   -I"C:\eclipse\arduinoPlugin\packages\STM32\hardware\stm32\1.5.0\cores\arduino" -I"C:\eclipse\arduinoPlugin\packages\STM32\hardware\stm32\1.5.0\variants\BLUEPILL_F103XX" -MMD -MP -MF"core\core\stm32\LL\stm32yyxx_ll_comp.c.d" -MT"core\core\stm32\LL\stm32yyxx_ll_comp.c.o" -D__IN_ECLIPSE__=1 "C:\eclipse\arduinoPlugin\packages\STM32\hardware\stm32\1.5.0\cores\arduin [... cut ...]
' '
'Finished building: C:\eclipse\arduinoPlugin\packages\STM32\hardware\stm32\1.5.0\cores\arduino\stm32\LL\stm32yyxx_ll_comp.c'
' '
'Finished building: C:\eclipse\arduinoPlugin\packages\STM32\hardware\stm32\1.5.0\cores\arduino\stm32\LL\stm32yyxx_ll_crc.c'
' '
makefile:71: recipe for target 'all' failed
make: *** [all] Error 2
"C:/eclipse/arduinoPlugin/tools/make/make -j4 all" terminated with exit code 2. Build might be incomplete.

20:29:35 Build Failed. 13 errors, 0 warnings. (took 7s.176ms)`

fpistm, 2019-03-28:
Hi @cdrose 
right some people already raised this issue.
I never met him.
Anyway, with Arduino IDE (on Windows and Linux), `USB_PRODUCT` is well defined:
`"-DUSB_PRODUCT=\"BLUEPILL_F103C8\""`
With Eclipse + Sloeber on Linux, it works also:
`'-DUSB_PRODUCT="BLUEPILL_F103C8"'`

I could not test under windows as I get a proxy issue to install Sloeber.
Anyway on you log we see that:
`"-DUSB_PRODUCT="BLUEPILL_F103C8""`

This is why you get this error. Don't know how Sloeber parse the platform.txt but it is linked to that. I advise you to submit an issue on Sloeber.
In platform.txt this is how it is defined:
`'-DUSB_PRODUCT="{build.board}"'`
https://github.com/stm32duino/Arduino_Core_STM32/blob/9299ab2708bf9db9f18957dd2a921184079ecd19/platform.txt#L70

One workaround could be to edit the properties of your project, then go to "_C/C++ Build_" -> "_Environment_" and search "`A.BUILD.USB_FLAGS`"
Then you can edit it to fix the USB_PRODUCT definition with Sloeber.
On Linux I get:
`-DUSBCON ${A.BUILD.USB_SPEED} -DUSBD_VID=${A.BUILD.VID} '-DUSB_MANUFACTURER=${A.BUILD.USB_MANUFACTURER}' '-DUSB_PRODUCT="${A.BUILD.BOARD}"' -DHAL_PCD_MODULE_ENABLED`

fpistm, 2019-03-29:
Any feedback ?

cdrose, 2019-03-29:
Sorry haven't been able to look into this further yet. Ill see if anyone has mentioned it on the Sloeber page.

cdrose, 2019-03-31:
It looks like this has been raised previously with Sloeber and this fix is non-trivial:
https://github.com/Sloeber/arduino-eclipse-plugin/issues/1047#issuecomment-443458088

fpistm, 2019-04-03:
Thanks @cdrose 
I close this one as linked to how Sloeber handle/parse the platform.txt.

closed 2019-04-03 by fpistm