stm32duino/Arduino_Core_STM32 #487: could'nt recieve integers at master from slave

DeadSHoT555, 2019-03-29:
well in examples the character array works pretty good..., but when modified to send integer there is no serial print ... gets struck at some unknown point

fpistm, 2019-03-29:
@DeadSHoT555 
As mention in your preview issue opened in BoardManagerFiles repo, could you provide more information.
Host OS, IDE version, core version, sketch you used, what is expected...
There an issue template for that, so pleas follow it else I will not be able to provide support.
https://github.com/stm32duino/Arduino_Core_STM32/blob/master/.github/ISSUE_TEMPLATE/bug_report.md

fpistm, 2019-03-29:
Well, you opened a second one... #488.
You can edit your issue.

closed 2019-03-29 by fpistm

DeadSHoT555, 2019-04-02:
i tried with a new pair of boards still could'nt get numerical bytes...can anyone verify the problem