stm32duino/Arduino_Core_STM32 #488: could'nt recieve integers at master from slave

DeadSHoT555, 2019-03-29:
well in 'Wire examples' for (stm32f10c bluepill) the character array works pretty good..., but when slave sender modified to send integer there is no serial print in master reader ... gets struck at some unknown point...,  works fine for one byte transfer....,and not any more for two bytes or more.. can anyone fix this bug???

fpistm, 2019-03-29:
@DeadSHoT555  could you provide more information.
Host OS, IDE version, core version, exact sketches you used, what is expected...
There an issue template for that, so pleas follow it else I will not be able to provide support.
https://github.com/stm32duino/Arduino_Core_STM32/blob/master/.github/ISSUE_TEMPLATE/bug_report.md

DeadSHoT555, 2019-03-29:
OS: [ Windows10]
Arduino IDE version: [1.8.9]
STM32 core version: [1.5.0]
Tools menu settings [Newlib nano(default)]
Upload method: [serial]
Board (Generic STM32F1 Series):
Name: [BluePill F103C8]

DeadSHoT555, 2019-03-29:
// Wire Master Reader
// by Nicholas Zambetti <http://www.zambetti.com>

// Demonstrates use of the Wire library
// Reads data from an I2C/TWI slave device
// Refer to the "Wire Slave Sender" example for use with this

// Created 29 March 2006

// This example code is in the public domain.

#include <Wire.h>

void setup()
{
  Wire.begin();        // join i2c bus (address optional for master)
  Serial.begin(9600);  // start serial for output
}

void loop()
{
  Wire.requestFrom(8, 2);    // request 6 bytes from slave device #2

//  while(Wire.available())    // slave may send less than requested
//  { 
    byte c = Wire.read(); // receive a byte as character
    byte d = Wire.read(); // receive a byte as character
    Serial.print(c);         // print the character
//  }

  delay(500);
}

### master_sender

DeadSHoT555, 2019-03-29:
// Wire Slave Sender
// by Nicholas Zambetti <http://www.zambetti.com>

// Demonstrates use of the Wire library
// Sends data as an I2C/TWI slave device
// Refer to the "Wire Master Reader" example for use with this

// Created 29 March 2006

// This example code is in the public domain.

#include <Wire.h>

void setup() {
  Wire.begin(8);                // join i2c bus with address #8
  Wire.onRequest(requestEvent); // register event
}

void loop() {
  delay(100);
}

// function that executes whenever data is requested by master
// this function is registered as an event, see setup()
void requestEvent() {
  Wire.write(22); // respond with message of 6 bytes
  Wire.write(33); // respond with message of 6 bytes
  // as expected by master
}
### slave_sender

DeadSHoT555, 2019-03-29:
please ignore the comments there.... i have just modified the example code

DeadSHoT555, 2019-03-29:
[image]

fpistm, 2019-03-29:
Please kept original code then update it to send/receive only one integer.
Note that integer will be cast to uint8_t so ensure it is less than 255.

DeadSHoT555, 2019-03-29:
yeah we are able to transfer master to slave by multiple bytes(as you said unit8_t) but unable to transfer from slave to master using same bytes, it is supporting only one byte transfer. i could'nt transfer two bytes or more....

DeadSHoT555, 2019-03-29:
the prescise problem is like to transfer  two or more bytes from slave to master

DeadSHoT555, 2019-03-29:
note: issue only on numerical bytes... character bytes are going good

fpistm, 2019-03-29:
Transferring a char, a byte or a casted value to uint8_t is the same.
Your code  stuck because you do not handle properly how you request and how you send. Then I2C is probably in busy state as it wait.
Even your screenshot code of both ketch is not aligned.
Using available API is the right way to ensure they are something to read:
https://www.arduino.cc/en/Reference/WireAvailable

Further reading: Wire API
https://www.arduino.cc/en/Reference/Wire

DeadSHoT555, 2019-03-29:
i have communicated between multiple arduino'susing i2c both ways and stm32 series with roger clark's stm32 addon... and also here in one way but there is bug  for the other way..(.) i do too agree that char byte and numerical byte is same but the char on moodification in any way it is working but for numerical byte transfer only one byte is getting transferred, if i do request 2 or more bytes it strucks.. and when ever i reset the board it it prints once only the last byte not any other bytes.        i tried most possible ways of i2c basic communication examples available online for 6 hours,  none resulted luck. and last the screenshot was a mistaken one. one can consider those two codes are the real challange of this issue#488 master as promini and slave as stm32

rikykumar, 2019-04-02:
if you make stm32 as a master and promini as slave it is working fine in both ways . but when stm32 was made to be  slave and promini as master.. could not get numerical data from promini to stm32
really fishy

DeadSHoT555, 2019-04-02:
i tried with a new pair of boards still could'nt get numerical bytes...can anyone verify the problem

DeadSHoT555, 2019-04-02:
Frederic pillion,
can u once look into this issue personnaly.. because it one of the basic requirement with i2c with stm32.... waiting for ur reply

fpistm, 2019-04-02:
Of course this issue will be manage but don't know when.

poojaasritha, 2019-04-02:
[image]
[image]

poojaasritha, 2019-04-02:
yup!
 the problem is true

fpistm, 2019-04-02:
Maybe linked to https://github.com/stm32duino/Arduino_Core_STM32/issues/472

rikykumar, 2019-04-02:
> Maybe linked to #472

but for one byte transfer it is working good ... and if there are two or more bytes it is getting problem so can it be a interrupt problem.., i think it is problem with onRequest handler

DeadSHoT555, 2019-04-02:
that too could'nt  solve the problem... and there is a percular property that can be seen in the screenshot. whenever i press reset button on stm32 there used to be a last byte transfer only once and stops not another byte...and if i press very fast i used to  get -1 continously

DeadSHoT555, 2019-04-02:
[image]

LMESTM, 2019-04-02:
what if you try below code?
```
Wire.write(0x1234);
```
or 
```
Wire.write("12");
```

DeadSHoT555, 2019-04-02:
@rikykumar could'nt get data from STM32 Slave to PROMINI Master

DeadSHoT555, 2019-04-02:
yeah! anything that is in inverted commas("  ") is getting transferred. Now the question is how to transfer the variables? [IMG_20190402_222726_689](https://user-images.githubusercontent.com/39310462/55421403-c7f9c100-5596-11e9-9dab-40f2b4452f66.jpg)
[image]
[image]

DeadSHoT555, 2019-04-02:
yeah! anything that is in inverted commas("  ") is getting transferred. Now the question is how to transfer the variables?
[image]
[image]
[image]

uzi18, 2019-04-02:
@DeadSHoT555 have a look on this: https://github.com/madsci1016/Arduino-EasyTransfer/tree/master/EasyTransferI2C

LMESTM, 2019-04-03:
And have you tried as well : Wire.write(0x1234); ?
I think that the key point is that you need a defined "protocol" in place between the master and the slave. The protocol defines a known size N for the transfer.
Master will request 
```Wire.RequestFrom(8, N);```
 and the slave should respond to the master with the N number of bytes with a single call to 
```Wire.write( parameter of N bytes);```

DeadSHoT555, 2019-04-03:
[image]
[image]

DeadSHoT555, 2019-04-03:
yeah but if promini Master asks for one byte ..., if promini Master asks two or more bytes no more transfer

DeadSHoT555, 2019-04-03:
yeah i'm checking that  "protocol"  all time, problem is not at all at that point .., those above screenshots are to represent its behaviour on different requests. if we do request 2 or more bytes it strucks

DeadSHoT555, 2019-04-03:
[image]
it is not getting compiled for stm32 series

poojaasritha, 2019-04-03:
in wire.onRequest(){ } after interrupt only one write is getting executed and that too the last byte

poojaasritha, 2019-04-03:
Laurent Meunier
finally i found that
in roger clarks<Wire_slave.h> slave is working very good not sure to make it to work in second wire. here in stm32duino second wire can be used but slave is not working. is there any possibility to debug slave error here fronm <Wire_slave.h>

uzi18, 2019-04-03:
what is your compilation error for easytransfer-arduino-library ?

LMESTM, 2019-04-03:
> in roger clarks<Wire_slave.h> slave is working very good not sure to make it to work in second wire. here in stm32duino second wire can be used but slave is not working. is there any possibility to debug slave error here fronm <Wire_slave.h>

Sorry but are only supporting stm32duino/Arduino_Core_STM32 here. The backlog of issues is already high so unfortunately we cannot afford investigating issues of Roger's core. Note that supporting other instances of peripherals mainly depends on the pin mapping and alternate functions, you may change the default Wire instance of Roger's core from instance 1 to instance 2 for your need ? 

Concerning our issue here, can you unblock your situation with using  
```Wire.write("12");```
?
you may put your variables in the string by using sprintf and alikes ...

LMESTM, 2019-04-03:
Ok I had a deeper look at Wire.h and discussed with @fpistm.
now my understanding it that

- For sending 1 byte only from slave to master, you can use 
    ```virtual` size_t write(uint8_t);```

- For sending more than 1 byte you MUST use:
   ``` virtual size_t write(const uint8_t *, size_t);```
which is the same as     size_t write(const char *str)   (write("") )

So you can create tables if you don't want to use string. Something like below should work
```
uint8_t variables[2];
variable[0] = 5;
variable[1] = 6;
Wire.write(variables, 2);
```
I agree that it would be nice to be able to make loops of  write(uint8_t); like promini would but that requires more investigations, this is not supported for now.

poojaasritha, 2019-04-04:
thanks a lot that's a great fix for now.., and i will be an active member in testing further developments involved in this topic

poojaasritha, 2019-04-04:
and sorry for the misconception there i mean to resolve bug in stm32duino i2c slave from <wire_slave.h>. any ways i'm glad that the problem is fixed

DeadSHoT555, 2019-04-04:
[image]
Yup finally its working

fpistm, 2019-04-04:
Is it ok between 2 pro mini ?
I tired to understand why it is functional when STM32 is master and pro mini slave. Regarding Wire implementation of the pro mini I don't understand why it is functional.
My guess is there is something in the HAL which is not correct (in pro mini or STM32).
Anyway, following the Arduino API, first proposed sketch is not correct from my point of view or I missed something 🤣

DeadSHoT555, 2019-04-06:
yeah 2 way communication between two prominis is good

closed 2019-09-12 by fpistm