stm32duino/Arduino_Core_STM32 #489: There is an error when I try to upload my program to the STM32L432KC

cloudtom, 2019-04-01:
Error while detecting libraries included by C:\Users\w\AppData\Local\Temp\arduino_build_540589\sketch\Blink.ino.cpp
C:\\Users\\w\\AppData\\Local\\Temp\\arduino_build_540589\\preproc\\ctags_target_for_gcc_minus_e.cpp"
arm-none-eabi-g++: error: CreateProcess: No such file or directory

exit status 1

fpistm, 2019-04-01:
Hi @cloudtom, how did you install the core?
It seems you do not have the arm gcc toolchains installed.

cloudtom, 2019-04-01:
I have already instal the core as the procedure in the Getting Started page.
[image]

fpistm, 2019-04-01:
Well hard to tell what's going wrong.
I've started to use the Arduino IDE 1.8.9 and do not see any issue.
Is it a fresh Arduino install? If yes could you try to start from scratch? or clean up your install.
You could also check if all packages are well installed (Core, Tools, CMSIS and arm none eabi).

cloudtom, 2019-04-01:
Could you tell me how to check if all packages are well installed. I don't know how to check because they are installed by Ardunio IDE.

fpistm, 2019-04-01:
https://www.arduino.cc/en/guide/environment#toc13
Extract:

> Support for third-party hardware can be added to the hardware directory of your sketchbook directory. Platforms installed there may include board definitions (which appear in the board menu), core libraries, bootloaders, and programmer definitions. 

Sketchbook path are in the preferences menu.
Other links:
https://www.arduino.cc/en/hacking/preferences

You could also try a standalone version of the Arduino IDE:
Download archive install (zip).
Extract it and create a folder name "portable" inside the Arduino directory. Then all stuff will be put in portable directory (sketch books, packages and libraries install, preferences)
All are explained here: https://www.arduino.cc/en/Guide/PortableIDE

fpistm, 2019-04-01:
Here details of where are the sources: https://github.com/stm32duino/wiki/wiki/Where-are-sources

cloudtom, 2019-04-02:
The problem was solved after I installed the standalone version of the Arduino IDE. Thank you for your help!

closed 2019-04-02 by cloudtom

fpistm, 2019-04-02:
@cloudtom welcome.
Anyway this should work on the default installed. Probably an Arduino IDE issue. (cache, issue during install,...)

cloudtom, 2019-04-02:
Yes, I think it is an Ardunio IDE issue, maybe it is because I was using the Windows Store version? Just now, I help my classmate to install the core on his computer. It is successful. And didn't meet any problem. And his Ardunio is default version.

fpistm, 2019-04-02:
Note that for your case it is simply to use the standalone version, as you can get it on an USB key then simply copy where you want on several computer.