stm32duino/Arduino_Core_STM32 #492: Question:  how can I determine the parameters for dfu-util to load .bin standalone

johncblacker, 2019-04-02:
I want to use the maple_loader as a standalone, external tool under ChibiStudio to flash my maple-mini after compiling a program with ChibiStudio.  After looking at a verbose output of an Arduino sketch load to a maple-mini that has the STM32duino bootloader, I noticed that there is a parameter: "1EAF:003" which I presume is the device identifier.  I see that this parameter is supplied at the time the maple_loader.bat file is called.  How is this parameter derived?  I'm presuming that it is not constant; therefore in order for me to use the maple_loader.bat in ChibiStudio, I need to be able to insert the correct parameter.  Can someone explain how I can determine the correct value for this parameter?
Thanks.

fpistm, 2019-04-02:
Hi  @johncblacker,
this core does not support (yet) the dfu upload method nor the bootloader.
You probably used the core of @rogerclarkmelbourne available here:
https://github.com/rogerclarkmelbourne/Arduino_STM32

Note that 1EAF:003 is the USB PID:VID.

closed 2019-04-03 by fpistm

xymopen, 2019-05-31:
Succeed programming [Bluepill](https://wiki.stm32duino.com/index.php?title=Blue_Pill "Blue Pill - STM32duino wiki") with [STM32duino-bootloader](https://github.com/rogerclarkmelbourne/STM32duino-bootloader "rogerclarkmelbourne/STM32duino-bootloader: Bootloader for STM32F103 boards, for use with the Arduino_STM32 repo and the Arduino IDE"). I'll leave my experience here if you're interested.

Basically we need to borrow upload tools and menus from [Arduino_STM32](https://github.com/rogerclarkmelbourne/Arduino_STM32 "rogerclarkmelbourne/Arduino_STM32: Arduino STM32. Hardware files to support STM32 boards, on Arduino IDE 1.8.x including LeafLabs Maple and other generic STM32F103 boards") and modify the linker scripts.

First get the tools. I myself install the Arduino_STM32 cores and Arduino IDE can automatically find them even on another core.

Then make a copy of linker script (in my case it is `variants/BLUEPILL_F103XX/ldscript.ld`) and rename it to `bootloader_20.ld`. Change `MEMORY` section according to [bootloader_20.ld](https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/677de8c167379519fee793ce352b7e7bd0ebdf27/STM32F1/variants/generic_stm32f103c/ld/bootloader_20.ld) from Arduino_STM32 core. We will use the name later.

```diff
64c64
< FLASH (rx)      : ORIGIN = 0x8000000, LENGTH = LD_MAX_SIZE
---
> FLASH (rx)      : ORIGIN = 0x08002000, LENGTH = LD_MAX_SIZE
```

Finally place the borrowed menus. Create `boards.local.txt` and `platform.local.txt` under STM32 core folder.

<details><summary>boards.local.txt</summary>

```
GenF1.menu.upload_method.DFUUploadMethod=STM32duino bootloader
GenF1.menu.upload_method.DFUUploadMethod.upload.protocol=maple_dfu
GenF1.menu.upload_method.DFUUploadMethod.upload.tool=maple_upload
GenF1.menu.upload_method.DFUUploadMethod.upload.usbID=1EAF:0003
GenF1.menu.upload_method.DFUUploadMethod.upload.altID=2
# Note we need to define VECT_TAB_ADDR as VECT_TAB_OFFSET
GenF1.menu.upload_method.DFUUploadMethod.build.extra_flags=-D{build.product_line} {build.enable_usb} {build.xSerial} -DVECT_TAB_OFFSET=0x08002000
GenF1.menu.upload_method.DFUUploadMethod.build.ldscript=bootloader_20.ld

```

</details>

<details><summary>platform.local.txt</summary>

```
# Upload using Maple bootloader over DFU
tools.maple_upload.cmd=maple_upload
tools.maple_upload.cmd.windows=maple_upload.bat
tools.maple_upload.path={runtime.tools.stm32tools.path}/win
tools.maple_upload.path.macosx={runtime.tools.stm32tools.path}/macosx
tools.maple_upload.path.linux={runtime.tools.stm32tools.path}/linux
tools.maple_upload.path.linux64={runtime.tools.stm32tools.path}/linux
tools.maple_upload.upload.params.verbose=-d
tools.maple_upload.upload.params.quiet=
tools.maple_upload.upload.pattern="{path}/{cmd}" {serial.port.file} {upload.altID} {upload.usbID} "{build.path}/{build.project_name}.bin"

```

</details>

They are modified from [boards.txt](https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/677de8c167379519fee793ce352b7e7bd0ebdf27/STM32F1/boards.txt#L361-L396 "Arduino_STM32/boards.txt at master · rogerclarkmelbourne/Arduino_STM32") and [platform.txt](https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/677de8c167379519fee793ce352b7e7bd0ebdf27/STM32F1/platform.txt#L109-L177 "Arduino_STM32/platform.txt at master · rogerclarkmelbourne/Arduino_STM32"). We create `boards.local.txt` and `platform.local.txt` because we don' t want to mess `boards.txt` and `platform.txt`.

Now I can select `STM32duino bootloader` in the `Upload method` and upload the sketch. Note auto reset is not implement in STM32 core so we need to reset it manually in order to upload.

I hope ST may officially support more upload methods, namely `STM32duino bootloader` and `JLink`, in the future.

BennehBoy, 2019-05-31:
I'm not sure why you have done this, support for the stm32duino bootloader is already in the dev branch of the core see pr #525

xymopen, 2019-05-31:
Sorry, didn't see that.

fpistm, 2019-05-31:
No worries.
This will be available thanks the board manager with the next release 1.6.0.