stm32duino/Arduino_Core_STM32 #493: The analog signal change is very small when I use the A0 pin to read analog signal from a microphone.

cloudtom, 2019-04-03:
When I use STM32L432KC to read analog signal from a microphone, I clapped my hand but the change of the value is too small and even can't see any change in some situations. If and only if I knock the microphone, the value(from the serial monitor) will change. 
However when I used the mbed online IDE (https://ide.mbed.com/compiler/), I didn't found this problem. It can be detected when I clapped my hand normally.

This is my Arduino Program Code:
int micPin = A0; // microphone sensor input
int ledPin = D11; // select the pin for the LED 
int micValue = 0; // variable to store the value coming from the mic sensor

void setup() { Serial.begin(9600);
pinMode(ledPin, OUTPUT);
}

void loop() { // read the value from the sensor:

micValue = analogRead(micPin);

micValue = constrain(micValue, 0, 100); /set sound detect clamp 0-100

Serial.print("incoming value from microphone =");

Serial.println( micValue);

if (micValue > 48){
digitalWrite(ledPin, HIGH);}else{
  digitalWrite(ledPin, LOW);
}
delay(100); }
If this code run succcessfully, the LED will be on when clapped my hands, but in the real situation it doesn't have any change when I clapped my hands. Only when I hit the microphone the LED will be on. Which means the sound needs to be very very large. And from the serial monitor I can saw that the analog value didn't change when I clapped my hands or speak to the microphone. I also try to change the minimum value(micValue > 48) to some smaller value such as 47. When I change the value to 47  the LED light will always on. I can't find a suitable value. I think sensitivity is not enought.

This is the mbed online IDE code:
#include "mbed.h"
AnalogIn A(A0);
DigitalOut led(D11);
int clapcount=0;

int main() {
    while(1) {
        int sound=A;
        if(clapcount>4){clapcount=1;}
        if(sound>0.1){
            clapcount++;
            wait(0.1);
            }
        if(clapcount==2){led=1;} 
        if(clapcount==4){led=0;}    
    }
}
When I use the mbed IDE it work well.
I will appreciate your help with this problem.

fpistm, 2019-04-03:
Well for mbed you `wait `only when you detect the clap while for Arduino you `delay ` each loop.
Moreover you print each value and you constrain it which led to increase the delay between each read. So, between each effective `analogRead`, the clap is probably missed.

Could you try this:
```
int ledPin = D11; // select the pin for the LED
int micValue = 0; // variable to store the value coming from the mic sensor

void setup() {
  pinMode(ledPin, OUTPUT);
}

void loop() {
  micValue = analogRead(A0);
  if (micValue > 512) {
    digitalWrite(ledPin , HIGH);
    delay(100);
  } else {
    digitalWrite(ledPin , LOW);
  }
  delay(1);
}
```

or to be more precise:
```
int ledPin = D11; // select the pin for the LED

void setup() {
  pinMode(ledPin, OUTPUT);
}

void loop() {
  digitalWrite(ledPin , (analogRead(A0) > 512)  ? HIGH : LOW);
  delay(1);
}
```

Note that you can replace D11 by `LED_BUILTIN `as Nucleo L432KC has a user led.

cloudtom, 2019-04-04:
It works! Thanks a lot. You are so kind.

closed 2019-04-04 by cloudtom