stm32duino/Arduino_Core_STM32 #497: [New"Serial" and "Serial1" both  were declared on  same pins.. how to use "Serial" and "Serial1" lines individually Variant]

DeadSHoT555, 2019-04-12:
**Board description (please complete the following information):**
 - Name: [e.g. DISCOVERY F769NI]
 - Link to all relevant information:
   * Documentations
   * Schematics
   * Where this hardware can be purchased
  * ...

**Note that user can add a STM32 based board following this [wiki](https://github.com/stm32duino/wiki/wiki/Add-a-new-variant-(board))**

Any contribution is welcome, so do not hesitate to submit a PR.

DeadSHoT555, 2019-04-12:
[image]
when i am using Serial1 it is printing on the monitor

DeadSHoT555, 2019-04-12:
[image]
when i am using Serial  again it is prining on serial monitor

DeadSHoT555, 2019-04-12:
[image]
[image]
when i am using both the last declared one is getting active.hereby it is concluded that both "Serial" and "Serial1" pins were declared on same pins

fpistm, 2019-04-12:
Right, this is normal.
Serial is just a define on the default Serialx to use or SerialUSB

fpistm, 2019-04-12:
Why new variant?

DeadSHoT555, 2019-04-12:
in the mean while i want to communicate with gps with "serial1"  and serialMonitor with "Serial" then how to declare two individual commn lines
OS: [ Windows10]
Arduino IDE version: [1.8.9]
STM32 core version: [1.5.0]
Tools menu settings [Newlib nano(default)]
Upload method: [serial]
Board (Generic STM32F1 Series):
Name: [BluePill F103C8]

fpistm, 2019-04-12:
This is linked to how you wire your hardware.

DeadSHoT555, 2019-04-12:
[image]
as per the picture "Serial"  TX and RX is on A9 and A10  and  "Serial1"   TX and RX  is on A2 AND A3

DeadSHoT555, 2019-04-12:
thank you 
i got it to work now

fpistm, 2019-04-12:
`USART1` is on A9/A10 ans `USART2` on A2/A3.
Then by default `Serial1` is instantiate and `Serial` mapped to it and use `USART1`.
Then to use `Serial2`, it has to be instantiate using `-DENABLE_HWSERIAL2` or `HardwareSerial Serial2(PA3, PA2);`

Note:
`Serial` is the generic serial instance name.
`Serialx` is only a name, you can use the name you want.

For example, in your case, if you connect GPS on the `USART2`:
`HardwareSerial SerialGPS(PA3, PA2);`
And then use default `Serial` for monitor.

closed 2019-04-12 by fpistm