stm32duino/Arduino_Core_STM32 #498: Can't register Core callback

taskmgr, 2019-04-14:
The forum(http://www.stm32duino.com/) disabled the register features so I  post my problem here 

When I am trying to call the function 'registerCoreCallback()',It show me an error:

'registerCoreCallback' was not declared in this scope

Here's my code:

`void cb(){
  digitalWrite(PC13,HIGH);
delay(500);
digitalWrite(PC13,LOW);
delay(500);
  }
void setup()
{
 pinMode(PC13,OUTPUT);
  registerCoreCallback(&cb);
}
……
`
Even if I add #include <core_callback.h> the problem still exists.
I configured my Arduino IDE under the construction of this wiki 
https://github.com/stm32duino/wiki/wiki/Getting-Started

What should I do?:(

fpistm, 2019-04-14:
By default, it is disabled.
You have to enable it
https://github.com/stm32duino/Arduino_Core_STM32/blob/49eb0b80c75fc07ba21ff6c7b328c3ef6a89ffd7/cores/arduino/stm32/core_callback.h#L41

Use the `build_opt.h` to enable it:

https://github.com/stm32duino/wiki/wiki/Customize-build-options-using-build_opt.h

fpistm, 2019-04-15:
Wiki updated:
https://github.com/stm32duino/wiki/wiki/API#core-callback

closed 2019-04-15 by fpistm

taskmgr, 2019-04-16:
> 
> 
> Wiki updated:
> https://github.com/stm32duino/wiki/wiki/API#core-callback

Thanks