stm32duino/Arduino_Core_STM32 #505: confusing differences between this project and stm32duino official website

taskmgr, 2019-04-23:
I've read this wiki and find there are lots of differences between this wiki and this github project
http://wiki.stm32duino.com/index.php?title=API

> PWM
> .......
>For backward compatibility, in STM32duino, analogWrite() still takes values between 0 (always off) and 255 (always on); the new pwmWrite() function provides instead support to the higher resolution 0 (always off) and 65535 (always on).
Pin access
Pin Modes
 .......
    PWM: This is a special mode for when the pin will be used for PWM output (a special case of digital output).
......
>
For example,the function'pwmWrite()' is no longer available now;and the Pin Modes 'PWM' of 'pinMode()' function seems to be removed.

It's confusing...and I can't find more document about the differences compared to Arduino

fpistm, 2019-04-23:
Hi @taskmgr 
This is a different core. There is no legacy between those 2 cores.
This core is compatible with standard Arduino API (asmost possible as it can be, I hope 😆 )
So, default Arduino sketches can be used. For some API changes, update are referenced here:
https://github.com/stm32duino/wiki/wiki/API

This wiki http://wiki.stm32duino.com/index.php is for @rogerclarkmelbourne core.

More details are available here:
http://stm32duino.com/viewtopic.php?f=2&t=3111

taskmgr, 2019-04-24:
haha~Thanks for replying~Full of expectation for your work~

closed 2019-04-24 by taskmgr