stm32duino/Arduino_Core_STM32 #506: USB RNDIS

pauluzs, 2019-04-23:
Would it be possible to add a usb RNDIS device, So we can do networking over USB?
ie. run a udp server etc, Currently cores/arduino/stm32/usb/ contains only CDC and HID, 

Windows 10 should support rndis without drivers needed for
USB_DESC_TYPE_DEVICE,       /* bDescriptorType */
  0x00,                       /* bcdUSB */
  0x02,
  0xEF,                       /* bDeviceClass */
  0x04,                       /* bDeviceSubClass */
  0x01,                       /* bDeviceProtocol */

https://github.com/fetisov/lrndis and https://github.com/Lexatagan/LRNDIS
are 2 examples only not based on stm32duino core.

Modifying these device descriptors was one thing, 
Only implementing the endpoints for this is unfortunately above my knowledge.

Any help, links or useful info is welcome

fpistm, 2019-04-24:
Hi @pauluzs 
This probably could be achieved.
Quickly looking your link and see that all components are available (HAL, LWIP,...)
Currently, I will not have time to work on this but any help are welcome.

pauluzs, 2019-04-24:
Tnx @fpistm,
Been trying to understand the usb part in this core first before the LWIP. 
To add a new device: are all those files in the cores/arduino/stm32/usb/ folder? 
So i need to create the necessary RNDIS files and folders using cdc and hid as guidelines.
Or does this require more like having a custom rndis middleware class in the STM32_USB_Device_Library as well?

fpistm, 2019-04-24:
Yes all related USB stub are here: `cores/arduino/stm32/usb/`  and linked to USB middleware `STM32_USB_Device_Library`.
As far as I know there is no RNDIS class in the middleware, only those one are provided:
* Audio
* CDC
* CustomHID
* HID
*MSC

so I guess this is what the link you provide do anyway the middleware provide a template class so this could probably be used as a starting point.

fpistm, 2019-10-08:
Track in #687

closed 2019-10-08 by fpistm