stm32duino/Arduino_Core_STM32 #509: 3 compile errors in the example sketch HardwareTimerPWMInput.in

mrguen, 2019-04-26:
Hello,

I am looking for a way to use TIMER INPUT and found this nice example sketch " HardwareTimerPWMInput.in" but there are a few compilation errors:

1) Streaming.h: No such file or directory

Then after removing this include:

2)
HardwareTimerPWMInput:28:9: error: no match for 'operator<<' (operand types are 'USBSerial' and 'const char [80]')

  Serial << "Example of the Timer Input Capture mode to measure PWM pulse width and period\n\n";

3)
HardwareTimerPWMInput:73:61: error: 'endl' was not declared in this scope

   Serial << ", period: " << Timer2.getCompare(TIMER_CH1) << endl;

Configuration:
Win 7 64 pro
Arduino IDE 1.8.9
Arduino_Core_STM32 1.5.0 updated via boards manager

fpistm, 2019-04-26:
Hi  @mrguen 
This is normal as this example is not for this core. It is for this one:
https://github.com/rogerclarkmelbourne/Arduino_STM32
which is different.

mrguen, 2019-04-26:
Well, I just downloaded the package from https://github.com/stm32duino/Arduino_Core_STM32 and it is where I found this sketch...

mrguen, 2019-04-26:
@fpistm Ok, there is still something that I don't get: why the rest of the code compiles? For example I thought there was no support for TIMER INPUT in  Arduino_Core_STM32 but this code compiles with TIMER_IC_INPUT_DEFAULT and it uses the predefined timer instances...

fpistm, 2019-04-26:
`TIMER_IC_INPUT_DEFAULT`  is not defined in this core.
This core uses the CMSIS definition and STM32 HAL/LL from STMicrolectronics.
You can see #473 (an example is available in #447).
To use it you need to use the GitHub repository as I do not yet release it yet.
Furthermore an HardwareTimer library will be provide later to handle more easily Timer features (#146)

mrguen, 2019-04-26:
Yes, but it does compile...

fpistm, 2019-04-26:
Maybe, but I could not tell you why.
You use a sketch which is not for this core so I could not help you on this.
As it seems you use also @rogerclarkmelbourne core, use this example with this core.

mrguen, 2019-04-26:
Ok, I was confused because the boards manager shows "STM32F1Boards (STM32duino.com)" so I thought It was refering to an Arduino_Core_STM32 board. In fact I am using Roger Clark Melbourne libraries.

fpistm, 2019-04-26:
Welcome.
So you should open an issue here:
https://github.com/rogerclarkmelbourne/Arduino_STM32/issues

fpistm, 2019-04-26:
Close this one as OP open https://github.com/rogerclarkmelbourne/Arduino_STM32/issues/626

closed 2019-04-26 by fpistm