stm32duino/Arduino_Core_STM32 #515: Choosing SPI modules with overlapping Pins

4ndreas, 2019-05-05:
I'm using all three SPIports on the "Black F407VE" board with different pins. This causes the problem that the SPI modules that are already used are not blocked and overwritten by the next SPI instance. 

```
SPIClass SPI_1(PA6 , PA7, PA8);
SPIClass SPI_2(PC3, PC2, PB13);
SPIClass SPI_3(PB5, PB4, PB3);

```
Also doing it by calling the set* Function and begin afterwards doesn't work

```
  SPI_1.setMOSI(PA6 );
  SPI_1.setMISO(PA7);
  SPI_1.setSCLK(PA8);  
  SPI_1.begin();

  SPI_2.setMOSI(PC3);
  SPI_2.setMISO(PC2);
  SPI_2.setSCLK(PB13);  
  SPI_2.begin();

  SPI_3.setMOSI(PB5);
  SPI_3.setMISO(PB4);
  SPI_3.setSCLK(PB3);  
  SPI_3.begin();
```

The problem here is that there are 2 ways this can be routed since SPI1 and SPI3 share the same pins. Since the SPI1 is first in the list SPI3 is never chosen ( I guess) 

form the **PeripheralPins.c**
```
//*** SPI ***

#ifdef HAL_SPI_MODULE_ENABLED
const PinMap PinMap_SPI_MOSI[] = {
  {PA_7,  SPI1, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF5_SPI1)},
  {PB_5,  SPI1, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF5_SPI1)},
  {PB_5,  SPI3, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF6_SPI3)},
  {PB_15, SPI2, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF5_SPI2)},
  {PC_3,  SPI2, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF5_SPI2)},
  {PC_12, SPI3, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF6_SPI3)},
  {NC,    NP,    0}
};
#endif

#ifdef HAL_SPI_MODULE_ENABLED
const PinMap PinMap_SPI_MISO[] = {
  {PA_6,  SPI1, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF5_SPI1)},
  {PB_4,  SPI1, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF5_SPI1)},
  {PB_4,  SPI3, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF6_SPI3)},
  {PB_14, SPI2, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF5_SPI2)},
  {PC_2,  SPI2, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF5_SPI2)},
  {PC_11, SPI3, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF6_SPI3)},
  {NC,    NP,    0}
};
#endif

#ifdef HAL_SPI_MODULE_ENABLED
const PinMap PinMap_SPI_SCLK[] = {
  {PA_5,  SPI1, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF5_SPI1)},
  {PB_3,  SPI1, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF5_SPI1)},
  {PB_3,  SPI3, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF6_SPI3)},
  {PB_10, SPI2, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF5_SPI2)},
  {PB_13, SPI2, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF5_SPI2)},
  {PC_10, SPI3, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF6_SPI3)},
  {NC,    NP,    0}
};
#endif
```

Right now I removed the second line in the pinmap and everything works fine, but i think this is not a long therm solution.

I recommend either a constructor where the user can force the SPI module used, or a global list of already used SPI modules witch blocks the future use and forces the "spi_init" function to chose a different module.

fpistm, 2019-05-06:
> The problem here is that there are 2 ways this can be routed since SPI1 and SPI3 share the same pins. Since the SPI1 is first in the list SPI3 is never chosen ( I guess)

You guess correctly.
I will comment it by default, I've missed this when variant was added.
About be able to manage this "dynamically" is not so easy and bring lot of code...
Pin/peripheral configuration is mainly linked to end user usage and or board schematics.
Anyway, one track I've add is to do like MBED, which defined PX_Y_ALTx definition.
This will allow to use it for init.
https://github.com/ARMmbed/mbed-os/blob/b21c278274d057ddfe3f8562497c7da922254d01/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F407xG/TARGET_DISCO_F407VG/PeripheralPins.c#L262
and https://github.com/ARMmbed/mbed-os/blob/b21c278274d057ddfe3f8562497c7da922254d01/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F407xG/TARGET_DISCO_F407VG/PinNames.h#L94

4ndreas, 2019-05-07:
Ok for now I think the custom board is a working solution. I like the "ALT0" approach. 
Thank you.

PR #555 (2019-07-01): Add WEAK attribute to PinMap arrays
`PinMap` arrays defined in `PeripheralPins.c` can now be overridden by user.
This allows to change peripheral used by a pin or save space if some peripherals are useless on pins.

Fix #515

This PR also fix duplicated pins definitions for BlackF407 raised in the same issue.

Updated default PinMap files:
https://github.com/stm32duino/Arduino_Tools/commit/4bd644e5c5eef6193e15a9ceaab898ff5955fb40

Wiki:
https://github.com/stm32duino/wiki/wiki/Custom-definitions#custom-pinmap-array
commit e10a8446 (2019-07-01): [PinMap] Add WEAK attribute to PinMap arrays

Allow custom overwriting.

Fix #515
closed 2019-07-01 by fpistm