stm32duino/Arduino_Core_STM32 #516: [EEXTR_F030_V1] Wrong HSE value?

fpistm, 2019-05-06:
Hi  @ghent360 
Regarding  EEXTR_F030_V1 schematics it seems HSE is 8MHz:
[image]

but in:
https://github.com/stm32duino/Arduino_Core_STM32/blob/5e2e32ab4f79ffbf9401eb2fc68ed403e974f145/variants/EEXTR_F030_V1/stm32f0xx_hal_conf.h#L86

what is the good value? `SystemClock_Config` should probably check also.

fpistm, 2019-05-06:
Seems 16000000is the good value as discussed about it here:
https://github.com/stm32duino/Arduino_Core_STM32/pull/397#discussion_r244095376
So schematics is not up to date, right?

ghent360, 2019-05-06:
> #397 (comment)

Correct. I was planning to use 8Mhz, but had a stack of 16MHz crystals
laying around, so ended up with those.

fpistm, 2019-05-06:
Ok. Thanks.

closed 2019-05-06 by fpistm