stm32duino/Arduino_Core_STM32 #517: Wrong pin gets toggled on Nucleo-144 F429ZI

sprucify, 2019-05-07:
I have a Nucleo 144 F429ZI board on top of a PCB I build.
The following simple blinking led code should drive pin D47 of the microcontroller, but it actually is toggling a relay I have on pin D44.
```
// the setup function runs once when you press reset or power the board
void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(PC_12, OUTPUT);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(PC_12, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);                       // wait for a second
  digitalWrite(PC_12, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);                       // wait for a second
}
```

I uninstalled and reinstalled the STM32 package via board manager.
But no effect.
What am I missing here?

I selected Nucle0-144 board group with F429ZI as board type.

fpistm, 2019-05-07:
D47 is equal to PC12 and is a pin number.
PC_12 is not the same, it's a piname which represent GPIO port|GPIO pin.

sprucify, 2019-05-08:
Thanks, it is solved

closed 2019-05-08 by sprucify

fpistm, 2019-05-08:
Welcome