stm32duino/Arduino_Core_STM32 #519: No I2C clock on F429ZI

sprucify, 2019-05-09:
I'm debugging at the moment why my I2c isn't working on the Nucleo144 F429ZI board.
So I tested out with the following I2C scanner:

```
#include <Wire.h>

void setup() {

  Serial.begin(9600);
  Wire.setSDA(PB9);
  Wire.setSCL(PB8);
  Wire.begin();
  Serial.println("\nI2C Scanner");
}

void loop() {
  byte error, address;
  int nDevices;

  Serial.println("Scanning...");

  nDevices = 0;
  for(address = 1; address < 127; address++) {
    // The i2c_scanner uses the return value of
    // the Write.endTransmisstion to see if
    // a device did acknowledge to the address.

    Wire.beginTransmission(address);
    error = Wire.endTransmission();
    
    if (error == 0) {
      Serial.print("I2C device found at address 0x");
      if (address < 16) 
        Serial.print("0");
      Serial.println(address, HEX);

      nDevices++;
    }
    else if (error == 4) {
      Serial.print("Unknown error at address 0x");
      if (address < 16) 
        Serial.print("0");
      Serial.println(address, HEX);
    }    
  }
  if (nDevices == 0)
    Serial.println("No I2C devices found");
  else
    Serial.println("done");

  delay(5000);           // wait 5 seconds for next scan
}
```

While this code is running I measure no clock or data with the scope on PB8 and PB9
I assume `stm32f4xx_hal_conf.h` file in C:\Users\<username>\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.4.0\variants\NUCLEO_F429ZIis used by default to configure all the pins and enabling of I2C module?

fpistm, 2019-05-09:
Hi @sprucify 
Right, this is the correct path.
First thing, by default Wire instance already use PB8/PB9 by default. So you do not need:
 ```
Wire.setSDA(PB9);
 Wire.setSCL(PB8);
```
Have you tested with the latest core version 1.5.0?
Have you wired a pull up resistor on each I2C line ?

sprucify, 2019-05-09:
Yes latest core 1.5.0 and I have two 1.5k pull ups.

fpistm, 2019-05-09:
OK I asked because the path you gave show 1.4.0.
I will test tomorrow as I don't have my board today.

sprucify, 2019-05-09:
I will uninstall and reinstall to check core version.
1.5k pull-ups should ok right?

fpistm, 2019-05-09:
I use 4.7k

sprucify, 2019-05-09:
To be sure I picked a new one from the shelf and it has exactly the same issue (no I2C clock or data)
When I try SPI program via STMCubeIDE I see a clock line with way too much noise and no data on data line. 

Am I missing a jumper or solder bridge?

fpistm, 2019-05-10:
Hi @sprucify 
As I think, it works.
I have no board customization. Default setup should be correct.
I do not see why you tested SPI while original issue is on I2C?
What is the I2C device wired on PB8/PB9 (D14/D15)? and how?

LMESTM, 2019-05-10:
I agree - maybe there is an issue with the scope ? or the GND ?

sprucify, 2019-05-10:
Sorry about the confusion, I was wondering if there was something wrong with the peripheral clock. That's why I tried SPI.

I have the pct2075gv temp sensor on the I2C bus with address 1001000

Checked the scope on FTDI cable, scope should be fine.
Also tested different ground pins on the board, the one in the corners, but also on CN header.

Do you mind sharing what revision number is on your microcontroller?
Hard to believe but it could be issues on the chip itself.
There is a ERRATE for the STM32F4 addressing several I2C issues

According to this topic it could be also something to do with the order of clock initialization:
https://community.st.com/s/question/0D50X00009XkZ9F/stm32f4-i2c-issues-solved

fpistm, 2019-05-11:
I will chech on monday.
About clock order. I doubt this is your issue. This is the same since a while for all series without any issue.
You could check/try all is done here:
https://github.com/stm32duino/Arduino_Core_STM32/blob/5e2e32ab4f79ffbf9401eb2fc68ed403e974f145/cores/arduino/stm32/twi.c#L93

fpistm, 2019-05-11:
Did you tried with hiher pull up?

uzi18, 2019-05-11:
maybe try first to toggle these pins and check if they works as gpios without i2c function?

sprucify, 2019-05-11:
I totally overlooked you have to solder and desolder several solder-bridges on the board to connect I2C to the CN9 header. 

PB9 and PB8 (I2C) are not connected to A4 and A5 (pin 9 and 11) on  
ST Zio connector CN9.

closed 2019-05-11 by sprucify

fpistm, 2019-05-11:
Ahhh. That's why it is important to tell how it is wired precisely. I've assumed you use the D14/D15 or the CN12 morpho.