stm32duino/Arduino_Core_STM32 #520: twi: why use non-blocking functions?

benwaffle, 2019-05-14:
For example, in `i2c_master_write`, you call `HAL_I2C_Master_Transmit_IT()` and then busy wait until `HAL_I2C_GetState() == HAL_I2C_STATE_READY`.  Why not just use `HAL_I2C_Master_Transmit()`, the blocking function?

fpistm, 2019-05-14:
IIRW, this was introduce to not block.
https://github.com/stm32duino/Arduino_Core_STM32/commit/2e9dc4dad770750996824ff4534673554a5d8b5c

VVESTM, 2019-05-14:
Yes, we introduce this in order to be able to use I2C and SPI in the same time. With the blocking functions, there was too much delay in the SPI.

fpistm, 2019-05-17:
Any update on this?

benwaffle, 2019-05-17:
Why would this make a difference? As long as the interrupt priorities are right, the SPI interrupts can occur while I2C is running. Do we know where the delays came from?

uzi18, 2019-05-17:
maybe we need just add setNonblocking(true)

VVESTM, 2019-05-20:
Trying to get back on this subject... This code has been introduced by #115 .
I made a mistake on my previous comment. The delay was introduced in I2C instead of SPI...

The use case was the following : 
    * on L475 IoT board, sensors are connected on I2C and BTLE on SPI (there is an IT on BTLE Rx)
    * transfer on going on I2C, in polling mode.
    * data available on BTLE, and so, an SPI transfer is launched on BTLE IT context.
    * I2C transfer is blocked waiting for SPI ends. 
	=> this was making an error on I2C and we reach a timeout.

By replacing the polling mode by IT mode on I2C transfer, we can be interrupted by SPI.

fpistm, 2019-05-29:
@benwaffle is @VVESTM  answer your question?
Thanks in advance

benwaffle, 2019-05-29:
How is the timeout check different? Whether in polling or interrupt mode, it calls `HAL_GetTick()` to get the current time, and that interrupt fires in every case (even if SPI or I2C is running). So why would it timeout in one case and not in another?

fpistm, 2019-05-31:
Well, in the original case, I2C was not in IT mode.
Then when the BTLE request a RX thanks an EXTI, the callback was called immediately. If this happen during the `HAL_I2C_Master_Transmit()`, then this cause I2C transfer to fail.
Moving to `HAL_I2C_Master_Transmit_IT()` allowed to not break the I2C transfer as I2C has higher prio than the EXTI. That's all.

fpistm, 2019-08-21:
Considered answered.

closed 2019-08-21 by fpistm