stm32duino/Arduino_Core_STM32 #523: How can i change VECT_TAB_OFFSET for use custom bootloader?

stnih, 2019-05-17:
Hi,
I am using Visual Micro for Arduino_Core_STM32 with STM32F4-DISCO1. I developed custom bootloader and i checked. It is working jump to 0x08010000. 
For application layer, VECT_TAB_OFFSET must be change before build. I tried a few ways but didn't change the result.
Disco.build.extra_flags=-D{build.product_line} {build.enable_usb} {build.xSerial} -DVECT_TAB_OFFSET=0x10000
Arduino15\packages\STM32\hardware\stm32\1.5.0\system\STM32F4xx\system_stm32f4xx.c **Line 110**

fpistm, 2019-05-17:
Right defining -DVECT_TAB_OFFSET=0x10000 should work.
With Arduino IDE this work. With Visual micro don't know.

This one is not used in the build:
`Arduino15\packages\STM32\hardware\stm32\1.5.0\system\Drivers\CMSIS\Device\ST\STM32F4xx\Source\Templates\system_stm32f4xx.c Line 109`

This one is used and as you can see `VECT_TAB_OFFSET` can be redefined.
https://github.com/stm32duino/Arduino_Core_STM32/blob/9f757b44954e60db665aa41ed904d5e8e0073bd4/system/STM32F4xx/system_stm32f4xx.c#L109

So my guess is that something is wrong on your build env with Visual micro.

stnih, 2019-05-17:
Yes, I tried it with arduino ide. It's works. Does not work with Visual Micro. I'll ask the Visual Micro form. Thanks

closed 2019-05-17 by stnih