stm32duino/Arduino_Core_STM32 #524: Custom Bootloader and Application with ArduinoIDE

stnih, 2019-05-17:
I have a STM32F4-DISC1. I want develop custom bootloader so i followed step by step under.

Screenshots and Codes : [https://www.dropbox.com/s/6b7g78y4xzu38z0/Screenshots_and_Codes.rar?dl=0](url)

1. Installed Arduino IDE 1.8.9

2. Installed Arduino_Core_STM32 with Boards Manager in Arduino IDE

3. I copied in "Arduino15\packages\STM32\hardware\stm32\1.5.0\variants" the DISCO_F407VG folder with its contents and created two new names with the folder DISCO_F407VG_App and DISCO_F407VG_Boot

4. My bootloader size is 64KB so, changed line 63 in DISCO_F407VG_Boot/ldscript.ld, FLASH (rx) : ORIGIN = 0x8000000, LENGTH = 64K

5. Changed line 63 in DISCO_F407VG_App/ldscript.ld, FLASH (rx) : ORIGIN = 0x8010000, LENGTH = 960K # (1024 - 64 = 960K)

6. Added DISCO_F407VG_App and DISCO_F407VG_Boot in boards.txt

7. Create new Project for Bootloader project (for test, blink green led and send data)

8. Build Bootloader project (compiled hex file copy to desktop)

9. Added code -DVECT_TAB_OFFSET=0x10000 in boards.txt next Arduino IDE Restart

9. Create new Project for Application project (for test, blink blue led and send data)

10. Build Application project (compiled hex file copy to desktop)

11. Erase all sectors with STM32 ST-LINK Utility.

12. Program respectively, Bootloader.hex and Application.hex with STM32 ST-LINK Utility. That is ok.

Green Led is blinking and data is sending. it's not problem but when i press button, only Green Led blinking stoped not working Blue Led.

The same bootloader software is running the test software I wrote in CubeMX. So the bootloader is jumping into the application field.

Do I need to do something about interrupt lines before the jump? Maybe the arduino is using other interrupt.

fpistm, 2019-05-17:
I will not be able to review it before 2 or 3 weeks.
Anyway, my first comprehension is you use Arduino to build and flash your bootloader? 64k is huge for a bootloader.
In fact you want 2 applications at same time?

You can see #525 how we manage some bootloaders.

xC0000005, 2019-05-21:
Are your files reversed? Application.ino looks like a bootloader, while bootloader looks like an application and even logs "from application."

fpistm, 2019-06-04:
Well, looking at your code. It seems your variant is correct. 
Maybe one thing, the `VECT_TAB_OFFSET` is define for all Disco so both application redefine it. Anyway it seems you take care of this in your step 8.
Maybe I will protect the jump function with `extern "C"`.
Finally, the system core clock config have to be the same as if you start from reset so maybe it can be an issue.

As this issue is a dedicated use case and not related to a dedicated core problem, I close it. Use the forum instead as it is now fully operational.
Thanks.

closed 2019-06-04 by fpistm

serkanc, 2019-10-17:
Can you share your project files again? Dropbox files are not available...