stm32duino/Arduino_Core_STM32 #527: When the USB is not connected and the COM port is not open, system crashes.

stnih, 2019-05-19:
When COM port is open ;
- LED_BLUE is blinking.
- Green led on when I press the button.

When COM port is close or USB unconnection ;
- LED_BLUE blinking several times and stops.
- Green led not working when I press the button.

**System crashes when the com port is not open. How can we deal with this?**

```
void setup()
{
	pinMode(LED_BLUE, OUTPUT);
	pinMode(LED_GREEN, OUTPUT);
	pinMode(USER_BTN, INPUT);
	
	SerialUSB.begin();
}

void loop()
{
	SerialUSB.println("Hello from app.");
	
	digitalWrite(LED_BLUE, !digitalRead(LED_BLUE));
		
	if (digitalRead(USER_BTN) == HIGH) {
		digitalWrite(LED_GREEN, HIGH);
	}
	else {
		digitalWrite(LED_GREEN, LOW);
	}
	
	delay(100);
}
```

fpistm, 2019-05-19:
H @stnih  Which board, host os, core version , build options,...

stnih, 2019-05-19:
Board : Discovery
Board part number : STM32F407G-DISC1
Host OS : Windows 10 
Version : 1.5.0
Arduino Version : 1.8.9

What is build options?

fpistm, 2019-05-20:
Hi @stnih 
As I though, it is already corrected on master thanks https://github.com/stm32duino/Arduino_Core_STM32/pull/444
I've tested with 1.5.0 and reproduced your issue. On top of the master it is now ok.
In fact, this is not really a system crash, only a blocking state.

closed 2019-05-27 by fpistm