stm32duino/Arduino_Core_STM32 #528: USB Mass Storage ?

Bob-the-Kuhn, 2019-05-21:
I'm just wondering if a USB Mass Storage implementation is on the roadmap?  

It would be really nice to stop moving the SD card between the PC and the STM32F board.

I've looked at some of the examples.  Definitely over my pay grade.

fpistm, 2019-05-21:
Hi @Bob-the-Kuhn 
currently not. Anyway this could be achieved using STM32_USB_Device_Library class MSC:
https://github.com/stm32duino/Arduino_Core_STM32/tree/master/system/Middlewares/ST/STM32_USB_Device_Library/Class/MSC

closed 2019-05-21 by Bob-the-Kuhn

zhscust, 2022-01-01:
> Hi @Bob-the-Kuhn currently not. Anyway this could be achieved using STM32_USB_Device_Library class MSC: https://github.com/stm32duino/Arduino_Core_STM32/tree/master/system/Middlewares/ST/STM32_USB_Device_Library/Class/MSC

@Frederic Pillon :- can you please share any example for the same ?