stm32duino/Arduino_Core_STM32 #529: How to control STM32F103RC's boot0 and reset pins through Arduino, and connect the DTR and RTS pins of communication chip‘s  ch340

firstnamenie, 2019-05-23:
How to control STM32F103RC's boot0 and reset pins through Arduino, and connect the DTR and RTS pins of communication chip‘s  ch340

fpistm, 2019-05-23:
Hi @firstnamenie 
I never use the ch340 chips.
What is your use case about the boot0 and reset? I guess this is to jump to the built-in STM32 bootloader ? or a dedicated one?
Which board you would use?

firstnamenie, 2019-05-23:
Thank you for your letter;
we use this chips for jump to the built-in STM32 bootloader , CH340 is a USB to TTL level chip.
When we load the program, we need to press the reset key to load. Can we solve this problem through Arduino program and  how to solve.  Thank you very much.

fpistm, 2019-05-23:
Currently, there is nothing to handle this in the core.
This is something I've in mind...
The idea is to have a dedicated sequence to initiate jump to system memory which will start the bootloader.
For example setting a specific speed could allow to request the jump to STM32 BL mode.
For STM32F103: 0x1FFFF000 
See Table 126. Bootloader device-dependent parameters [AN2606](https://www.st.com/content/ccc/resource/technical/document/application_note/b9/9b/16/3a/12/1e/40/0c/CD00167594.pdf/files/CD00167594.pdf/jcr:content/translations/en.CD00167594.pdf)
Some info available here:
https://www.youtube.com/watch?v=cvKC-4tCRgw

firstnamenie, 2019-05-23:
ok  Thank you very much.....

 https://www.youtube.com/watch?v=cvKC-4tCRgw

closed 2019-06-28 by fpistm