stm32duino/Arduino_Core_STM32 #537: STM32F072B's COM port not being seen by Arduino IDE

Bustean, 2019-06-06:
I've followed all the steps, i had it working several weeks ago, but now the Arduino IDE can't see the COM port to which the board is connected. The board is not bad, because i can still program it with Eclipse just fine.

fpistm, 2019-06-06:
Hi @Bustean 
Could you be more precise?
Which board? I guess the 32f072bdiscovery ?
Which core version ?
Which Arduino IDE version?
Which host OS ?
...

Bustean, 2019-06-06:
Yes, the board is the stm32f072b discovery.
I have Arduino 1.9.0 beta running on Windows 10.
As far as the core version goes, i have no clue what it is or where to find it, sorry for being a newbie.

Bustean, 2019-06-06:
I just tried using the non-beta IDE and it is working

fpistm, 2019-06-06:
Ok so this is linked to Arduino IDE :)
You can open an issue on Arduino GitHub.
There are already several issues opened for the 1.9.x beta:
https://github.com/arduino/Arduino/issues?q=is%3Aissue+is%3Aopen+label%3A%22IDE+1.9.x+Beta%22
For example, one linked to COM display:
https://github.com/arduino/Arduino/issues/8849

closed 2019-06-06 by fpistm