stm32duino/Arduino_Core_STM32 #538: Cant find STM32 Cores in the boards manager

paulxavier86, 2019-06-08:
I have tried to install the STM32 Cores in the arduino IDE. I added the link in the preferences but in boards manager cant find the stm32 cores. I tried in 1.8.7 and 1.8.9 arduino

fpistm, 2019-06-08:
Hi @paulxavier86 
The STM32 core is easy to find if you select `contributed` in the `Type` list.
[image]
or search `stm32` if you let `All`:
[image]

If you do not find it, then check the link you add:
https://github.com/stm32duino/BoardManagerFiles/raw/master/STM32/package_stm_index.json
If it does not work check your internet connection.

paulxavier86, 2019-06-09:
Thanks for the reply.
The error was real inspite  of all your suggestions. What worked was that i deleted the arduino15 folder from mydocuments and tried again.

now it works.

closed 2019-06-09 by paulxavier86

fpistm, 2019-06-09:
Yes, sometimes, some local files are corrupted. I don't know why but relies to Arduino IDE.
Fine you find a solution. Note that sometimes, only cleaning some files in the local arduino is enough instead of delete all.

poetlusint692, 2022-02-23:
Master!!
I found STM32 Core in the boards Manager but, it can't install. Why? Please.....

fpistm, 2022-02-23:
> Master!! I found STM32 Core in the boards Manager but, it can't install. Why? Please.....

Proxy issue? Hard to tell without more information