stm32duino/Arduino_Core_STM32 #545: STM32F401CC

poky99, 2019-06-21:
Good day.
I add a new board STM32F401CCU6 I have something that does not work.
My
boards.txt
# Black F401CUU
GenF4.menu.pnum.BLACK_F401CC=F401CUU STM32F401CUU
GenF4.menu.pnum.BLACK_F401CC.upload.maximum_size=262144
GenF4.menu.pnum.BLACK_F401CC.upload.maximum_data_size=65536
GenF4.menu.pnum.BLACK_F401CC.build.board=BLACK_F401CC
GenF4.menu.pnum.BLACK_F401CC.build.product_line=STM32F401xC
GenF4.menu.pnum.BLACK_F401CC.build.variant=BLACK_F401CC

variant.h
#ifndef _VARIANT_ARDUINO_STM32_
#define _VARIANT_ARDUINO_STM32_

#ifdef __cplusplus
extern "C" {
#endif // __cplusplus

/*----------------------------------------------------------------------------
 *        Pins
 *----------------------------------------------------------------------------*/

// Bluepill USB connector on the top, MCU side - Blackpill USB connector on bottom, MCU Side  (pins are reversed vertically for Arduino Ananlog pin correct sequence.
// Left Side
#define PB9  0
#define PB8  1
#define PB7  2
#define PB6  3
#define PB5  4
#define PB4  5
#define PB3  6
#define PA15 7
#define PA12 8  // USB DP
#define PA11 9  // USB DM
#define PA10 10
#define PA9  11
#define PA8  12
#define PB15 13
#define PB14 14
#define PB13 15
#define PB12 16 // LED Blackpill
// Right side
#define PC13 17 // LED Bluepill
#define PC14 18
#define PC15 19
#define PA0  20 // A0
#define PA1  21 // A1
#define PA2  22 // A2
#define PA3  23 // A3
#define PA4  24 // A4
#define PA5  25 // A5
#define PA6  26 // A6
#define PA7  27 // A7
#define PB0  28 // A8
#define PB1  29 // A9
#define PB10 30

// Other
#define PB2  31 // BOOT1
#define PA13 32 // SWDI0
#define PA14 33 // SWCLK

// This must be a literal
#define NUM_DIGITAL_PINS        34
// This must be a literal with a value less than or equal to to MAX_ANALOG_INPUTS
#define NUM_ANALOG_INPUTS       5
#define NUM_ANALOG_FIRST        10

// On-board LED pin number
#define LED_BUILTIN             PC13
// SPI Definitions
#define PIN_SPI_SS              PA4
#define PIN_SPI_MOSI            PA7
#define PIN_SPI_MISO            PA6
#define PIN_SPI_SCK             PA5

// I2C Definitions
#define PIN_WIRE_SDA            PB7
#define PIN_WIRE_SCL            PB6

// Timer Definitions
// Do not use timer used by PWM pins when possible. See PinMap_PWM.
#define TIMER_TONE              TIM3

// Do not use basic timer: OC is required
#define TIMER_SERVO             TIM2  //TODO: advanced-control timers don't work

// UART Definitions
#define SERIAL_UART_INSTANCE    1
// Default pin used for 'Serial' instance
// Mandatory for Firmata
#define PIN_SERIAL_RX           PA10
#define PIN_SERIAL_TX           PA9

/* HAL configuration */
#define HSE_VALUE               25000000U

/* Extra HAL modules */
//#define HAL_DAC_MODULE_ENABLED
//#define HAL_SD_MODULE_ENABLED

#ifdef __cplusplus
} // extern "C"
#endif
/*----------------------------------------------------------------------------
 *        Arduino objects - C++ only
 *----------------------------------------------------------------------------*/

#ifdef __cplusplus
// These serial port names are intended to allow libraries and architecture-neutral
// sketches to automatically default to the correct port name for a particular type
// of use.  For example, a GPS module would normally connect to SERIAL_PORT_HARDWARE_OPEN,
// the first hardware serial port whose RX/TX pins are not dedicated to another use.
//
// SERIAL_PORT_MONITOR        Port which normally prints to the Arduino Serial Monitor
//
// SERIAL_PORT_USBVIRTUAL     Port which is USB virtual serial
//
// SERIAL_PORT_LINUXBRIDGE    Port which connects to a Linux system via Bridge library
//
// SERIAL_PORT_HARDWARE       Hardware serial port, physical RX & TX pins.
//
// SERIAL_PORT_HARDWARE_OPEN  Hardware serial ports which are open for use.  Their RX & TX
//                            pins are NOT connected to anything by default.
#define SERIAL_PORT_MONITOR     Serial
#define SERIAL_PORT_HARDWARE    Serial
#endif

#endif /* _VARIANT_ARDUINO_STM32_ */

variant.cpp
#include "pins_arduino.h"

#ifdef __cplusplus
extern "C" {
#endif

// Pin number
const PinName digitalPin[] = {
  /* USB connector on the top, MCU side */
  /* Left Side */
  PB_9,  //D0
  PB_8,  //D1
  PB_7,  //D2
  PB_6,  //D3
  PB_5,  //D4
  PB_4,  //D5
  PB_3,  //D6
  PA_15, //D7
  PA_12, //D8 - USB DP
  PA_11, //D9 - USB DM
  PA_10, //D10
  PA_9,  //D11
  PA_8,  //D12
  PB_15, //D13
  PB_14, //D14
  PB_13, //D15
  PB_12, //D16 - LED Blackpill
  /* Right side */
  PC_13, //D17 - LED Bluepill
  PC_14, //D18
  PC_15, //D19
  PA_0,  //D20/A0
  PA_1,  //D21/A1
  PA_2,  //D22/A2
  PA_3,  //D23/A3
  PA_4,  //D24/A4
  PA_5,  //D25/A5
  PA_6,  //D26/A6
  PA_7,  //D27/A7
  PB_0,  //D28/A8
  PB_1,  //D29/A9
  PB_10, //D30
  PB_2,  //D31 - BOOT1
  PA_13, //D32 - SWDI0
  PA_14, //D33 - SWCLK
};

#ifdef __cplusplus
}
#endif

// ----------------------------------------------------------------------------

#ifdef __cplusplus
extern "C" {
#endif

WEAK void SystemClock_Config(void)
{
   RCC_OscInitTypeDef RCC_OscInitStruct;
  RCC_ClkInitTypeDef RCC_ClkInitStruct;

  /**Configure the main internal regulator output voltage
  */
  __HAL_RCC_PWR_CLK_ENABLE();

  __HAL_PWR_VOLTAGESCALING_CONFIG(PWR_REGULATOR_VOLTAGE_SCALE1);

  /**Initializes the CPU, AHB and APB busses clocks
  */
  RCC_OscInitStruct.OscillatorType = RCC_OSCILLATORTYPE_HSE;
  RCC_OscInitStruct.HSEState = RCC_HSE_ON;
  RCC_OscInitStruct.PLL.PLLState = RCC_PLL_ON;
  RCC_OscInitStruct.PLL.PLLSource = RCC_PLLSOURCE_HSE;
  RCC_OscInitStruct.PLL.PLLM = 25;
  RCC_OscInitStruct.PLL.PLLN = 168;
  RCC_OscInitStruct.PLL.PLLP = RCC_PLLP_DIV2;
  RCC_OscInitStruct.PLL.PLLQ = 4;
  if (HAL_RCC_OscConfig(&RCC_OscInitStruct) != HAL_OK) {
    _Error_Handler(__FILE__, __LINE__);
  }

  /**Initializes the CPU, AHB and APB busses clocks
  */
  RCC_ClkInitStruct.ClockType = RCC_CLOCKTYPE_HCLK | RCC_CLOCKTYPE_SYSCLK
                                | RCC_CLOCKTYPE_PCLK1 | RCC_CLOCKTYPE_PCLK2;
  RCC_ClkInitStruct.SYSCLKSource = RCC_SYSCLKSOURCE_PLLCLK;
  RCC_ClkInitStruct.AHBCLKDivider = RCC_SYSCLK_DIV2;
  RCC_ClkInitStruct.APB1CLKDivider = RCC_HCLK_DIV2;
  RCC_ClkInitStruct.APB2CLKDivider = RCC_HCLK_DIV1;

  if (HAL_RCC_ClockConfig(&RCC_ClkInitStruct, FLASH_LATENCY_5) != HAL_OK) {
    _Error_Handler(__FILE__, __LINE__);
  }
}

#ifdef __cplusplus
}
#endif

fpistm, 2019-06-21:
Hi @poky99 
could you add more info.
Host OS? Arduino IDE version?
Core version? Board infos (schelmatics, pinouts,...)?
And what exactly does not work?

poky99, 2019-06-21:
here is my post [http://www.stm32duino.com/viewtopic.php?f=39&t=4701&start=10#p54744](url)
 OS Win x64, IDE 1.8.9
Core 1.6
Even the Blink test doesn't work.
Checked in SystemWorkbench works, and took settings from it

fpistm, 2019-06-21:
Hard to tell without a schematics...
is the LED really on `PC13` ?
Is the `HSE` really 25MHz?
Maybe it crashes in one  `_Error_Handler` in  `SystemClock_Config`...
Best way is you try to debug.

poky99, 2019-06-21:
Yes 
LED = PC13
HSE = 25

fpistm, 2019-06-21:
Your clock config is not correct as the USB clock is 42 MHz instead of 48MHz, you could try this one to get it:
```C
  RCC_OscInitTypeDef RCC_OscInitStruct = {};
  RCC_ClkInitTypeDef RCC_ClkInitStruct = {};

  /** Configure the main internal regulator output voltage 
  */
  __HAL_RCC_PWR_CLK_ENABLE();
  __HAL_PWR_VOLTAGESCALING_CONFIG(PWR_REGULATOR_VOLTAGE_SCALE2);
  /** Initializes the CPU, AHB and APB busses clocks 
  */
  RCC_OscInitStruct.OscillatorType = RCC_OSCILLATORTYPE_HSE;
  RCC_OscInitStruct.HSEState = RCC_HSE_ON;
  RCC_OscInitStruct.PLL.PLLState = RCC_PLL_ON;
  RCC_OscInitStruct.PLL.PLLSource = RCC_PLLSOURCE_HSE;
  RCC_OscInitStruct.PLL.PLLM = 25;
  RCC_OscInitStruct.PLL.PLLN = 336;
  RCC_OscInitStruct.PLL.PLLP = RCC_PLLP_DIV4;
  RCC_OscInitStruct.PLL.PLLQ = 7;
  if (HAL_RCC_OscConfig(&RCC_OscInitStruct) != HAL_OK)
  {
    Error_Handler();
  }
  /** Initializes the CPU, AHB and APB busses clocks 
  */
  RCC_ClkInitStruct.ClockType = RCC_CLOCKTYPE_HCLK|RCC_CLOCKTYPE_SYSCLK
                              |RCC_CLOCKTYPE_PCLK1|RCC_CLOCKTYPE_PCLK2;
  RCC_ClkInitStruct.SYSCLKSource = RCC_SYSCLKSOURCE_PLLCLK;
  RCC_ClkInitStruct.AHBCLKDivider = RCC_SYSCLK_DIV1;
  RCC_ClkInitStruct.APB1CLKDivider = RCC_HCLK_DIV2;
  RCC_ClkInitStruct.APB2CLKDivider = RCC_HCLK_DIV1;

  if (HAL_RCC_ClockConfig(&RCC_ClkInitStruct, FLASH_LATENCY_2) != HAL_OK)
  {
    Error_Handler();
  }
```

fpistm, 2019-06-21:
```C
#define NUM_ANALOG_INPUTS       5
#define NUM_ANALOG_FIRST        10
```

Considering your comments this should be:
```C
#define NUM_ANALOG_INPUTS       10
#define NUM_ANALOG_FIRST        20
```

poky99, 2019-06-21:
Yes, I have installed
#define NUM_ANALOG_INPUTS       10
#define NUM_ANALOG_FIRST        20
Made changes but without result

fpistm, 2019-06-21:
Unfortunately, I will not be able to help more...
As I do not have the board, I could not perform any tests/debug.
You an try one of those method to debug:
https://github.com/stm32duino/wiki/wiki/How-to-debug

BennehBoy, 2019-06-21:
> Yes, I have installed
> #define NUM_ANALOG_INPUTS 10
> #define NUM_ANALOG_FIRST 20
> Made changes but without result

And you made the clock config changes?

poky99, 2019-06-21:
Yes

BennehBoy, 2019-06-21:
What about your PeripheralPins.c & PinNamesVar.h ?

If those are misconfigured - reuse in different modules etc could that not also cause issues @fpistm

poky99, 2019-06-21:
PeripheralPins.c & PinNamesVar.h from
[https://github.com/stm32duino/Arduino_Tools/tree/master/src/genpinmap/Arduino](url)

poky99, 2019-06-21:
Thank.
Forgot to replace ldscript.ld

closed 2019-06-21 by poky99

fpistm, 2019-06-21:
@BennehBoy I assumed it correct as it succeed to flash anyway seems not 🙄

BennehBoy, 2019-06-21:
Can you submit a PR so us folk don't have to rework this ourselves, I have these boards in the post to me 😄

poky99, 2019-06-21:
[401cc.zip](https://github.com/stm32duino/Arduino_Core_STM32/files/3314904/401cc.zip)

kgamecarter, 2019-06-24:
Will F401cc add to the 1.7.0 release?

fpistm, 2019-06-24:
> 
> 
> Will F401cc add to the 1.7.0 release?

If someone submit  a PR, it could.

BennehBoy, 2019-06-24:
I'll submit a PR as soon as my board is here and it can be tested -
otherwise anyone else should feel free to do so...

poky99, 2019-06-24:
I tested
I2C1 - work
I2C2 - work
I2C3 - work
USB CDC - work
RTC - work

BennehBoy, 2019-07-15:
My boards arrived, just looking at creating a variant for this and I notice that you didn't comment any of the alternate assignments out in peripheralpins.c - this I believe means that the last assigment will be the default ones?  @fpistm can you confirm?  Presumably there may be some timer conficts this way?

poky99, 2019-07-15:
> My boards arrived, just looking at creating a variant for this and I notice that you didn't comment any of the alternate assignments out in peripheralpins.c - this I believe means that the last [...]

I do not use remap, and so far there is no time to use this board in any project. I checked all I2C, UART and SPI.

BennehBoy, 2019-07-15:
Which UART's did you check?

USART2 & 6 won't initialise (6 conflicts with USB)

-DENABLE_HWSERIAL2 in build_opt.h does not result in a defined Serial2 device

poky99, 2019-07-15:
I'm sorry, I did not write a post correctly.
All I2C,
USART1
SPI1,
USB CDC

fpistm, 2019-07-15:
For array in `PeripheralPins.c` this is the first PinName found which will be used.

BennehBoy, 2019-07-15:
OK, I new it was one way round or the other, thanks for clarifying.