stm32duino/Arduino_Core_STM32 #547: Erasing EEPROM taking over 30 seconds

mayurharge, 2019-06-24:
I have an EEPROM reset function where I erase EEPROM completely.
the function is something like this.
 `{   DBG("erasing EEPROM.");
    unsigned long  start = millis();
    for (unsigned int i = 0; i < EEPROM_SIZE; i++) 
    {       
        EEPROM.write(i, 0x00);
        IWatchdog.reload();
    }
    DBG("Time took : ", millis() - start);
    DBG("Erased eeprom");
    // NVIC_SystemReset();
}`

> [602487] erasing EEPROM.

> [632579] Time took :  30092

> [632579] Erased eeprom.

to execute this function it's taking more than 30 seconds. 
please look into this.

fpistm, 2019-06-24:
Hi @mayurharge 
could you be more precise.
Which board, host OS, IDE version,...

mayurharge, 2019-06-25:
> Hi @mayurharge
> could you be more precise.
> Which board, host OS, IDE version,...
> 
> For reference, you should use EEPROM buffered access:
> #302
> 
> Here example:
> https://github.com/stm32duino/STM32Examples/blob/master/examples/NonReg/BufferedEEPROM/BufferedEEPROM.ino

Sorry for incomplete details, here is specific info
- Arduino IDE - 1.8.0
- OS - tested on both win10 & Mac OS
- Board - STM32F1 series (STM32F103Bt6 128kb version)

fpistm, 2019-06-25:
Thanks for info.

As mentioned in #302 by @hasenbanck 

> The current eeprom emulation didn't had any buffered access to the
> data on the flash. Every access to a byte would trigger a whole
> page read / write. For writes it would mean, that the whole page
> would be erased. For backward compatibility we will keep
> the current functions as they are, but will add functions
> to fill/flush the buffer and read from it.

This means with your code,  a loop of 1024,  will erase and write 1024 times the 1kb flash page.
So, as a write takes 30ms you get 30s for the total EEPROM size.

To avoid this @hasenbanck provides a way to work on buffered EEPROM and when need write only once:
```C
// Clear the EEPROM buffer
  for (uint16_t i = 0; i < EEPROM.length(); i++) {
    eeprom_buffered_write_byte(i, 0);
  }
  // Copy the data from the buffer to the flash
  eeprom_buffer_flush();
```

In this case this takes only 30ms to clear the flash.

mayurharge, 2019-06-26:
> Thanks for info.
>
> As mentioned in #302 by @hasenbanck
>
>
> This means with your code, a loop of 1024, will erase and write 1024 times the 1kb flash page.
> So, as a write takes 30ms you get 30s for the [...]

Thanks for the suggestion, implemented the same in my code.

closed 2019-06-26 by mayurharge