stm32duino/Arduino_Core_STM32 #549: [FR] Porting Marlin 2.0 firmware to the new STM32MP1 MCU series?

Gamester17, 2019-06-26:
Can I suggest that you help port Marlin 2.0 firmware to run the new STM32MP1 MPU series?
https://blog.st.com/stm32mp1-mpu-stm32mp157a-ev1-stm32mp157c-dk2/

Marlin is an open source firmware for RepRap and other fused deposition modelling (FDM) 3D printers based on the Arduino platform and there is already support for STM32 MCU series in Marlin 2.0 RC1 firmware.

I am only an end-user / technology enthusiast myself and I do not actually know if anyone is working on a 3D-printer controller board based on the new STM32MP1 MPU series or not, however, I read that STMicroelectronics claims STM32MP1 architecture enables developers to use the same software as STM32 MCU series so porting Marlin 2.0 firmware to STM32MP1 MPU series should be possible in theory as it already supports STM32, and theoretically, with the STM32MP1 MPU series you could run both Marlin and OctoPrint or similar Linux distribution on the same board.

For technical specification summery checkout:
https://www.cnx-software.com/2019/02/21/stmicro-stm32mp1-cortex-a7-m4-mpu/

STM32MP157A-DK1 and STM32MP157C-DK2 Discovery kits are $99 with LCD or $69 without
https://www.st.com/en/evaluation-tools/stm32mp157a-dk1.html
https://www.st.com/en/evaluation-tools/stm32mp157c-dk2.html
STM32MP157A-EV1 Evaluation kit for PCB engineers does have a steeper price at $399
https://www.st.com/en/evaluation-tools/stm32mp157a-ev1.html

First read about this MCU/MPU in the release news about PanGu board from I2Som
https://www.cnx-software.com/2019/06/25/pangu-board-stm32mp1-sbc/

fpistm, 2019-06-26:
Hi @Gamester17,

STM32MP1 is a MPU. This core is mainly for STM32 MCU.
I doubt Arduino core is the most appropriate way to support this.
An RTOS based framework like Zephyr or MBED would be more appropriate.

Gamester17, 2019-06-26:
@fpistm please read the linked specifications of STM32MP1 above as this MPU does have an integrated Cortex-M4 MCU @ 209MHz as well as one or two Cortex-A7 processors @ 650Mhz. That is, it combines the features of both a microcontroller as well as a microprocessor. STM32MP1 architecture enables developers to use the same software as STM32 MCU series (so compatible with software developed for STM32F1, STM32F3 and STM32F4 MCUs).

fpistm, 2019-06-26:
I know 😆 , what I would raised is that Cortex-A7 will not be supported.

fpistm, 2019-08-20:
STM32MP1 support is proposed by @kbumsik  here: #605

fpistm, 2019-10-28:
MP1 instroduction is ongoing, Cube package are now included and PR for board support has been sublitted.
Porting to Marlin have to be done in Marlin repo itself.

closed 2019-10-28 by fpistm