stm32duino/Arduino_Core_STM32 #551: I want to change frequency of STM32 pwm pin

gauri8427, 2019-06-28:
**Is your feature request/improvement related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context, example or screenshots about the feature request here.

Any contribution is welcome, so do not hesitate to submit a PR.

fpistm, 2019-06-28:
Hi  @gauri8427 
please use the forum for this kind of question as explained here : 
https://github.com/stm32duino/Arduino_Core_STM32#troubleshooting

_For question, support, ...,  you could submit a topic on the [stm32duino forum](http://stm32duino.com):_

 * _questions on the [STM32 Core](http://stm32duino.com/viewforum.php?f=48)_
 
 * _bugs/enhancements on the [STM core: Bugs and enhancements](http://stm32duino.com/viewforum.php?f=49)_ 

To answer your question:
https://github.com/stm32duino/wiki/wiki/API#frequency

fpistm, 2019-07-01:
Considered answered.

closed 2019-07-01 by fpistm

gauri8427, 2019-07-04:
But this didn't work for me

fpistm, 2019-07-04:
@gauri8427,
open a new topic on the forum.
As mention this is not the right place to ask support and be more precise about your config:
host OS, IDE version, core version, board used,...

sixtyfive, 2020-06-07:
This is the first hit on Google for "arduino stm32 pwm frequency" btw.