stm32duino/Arduino_Core_STM32 #556: STM32F103RET6

enesbcs, 2019-07-02:
**Board description:**
 - Name: STM32F103RET6 "Blue button" Development board
 - Link to all relevant information:
   * Documentations: https://github.com/mcauser/BLUE_BUTTON_F103RET6/
   * Where this hardware can be purchased: https://www.aliexpress.com/item/32787019266.html

"STM32F103xE" define can be found in some places in the Arduino_Core_STM32 but currently there are no board definitions for it.

BennehBoy, 2019-07-02:
It should be fairly simple to add this by following the guide -> https://github.com/stm32duino/wiki/wiki/Add-a-new-variant-(board)

Best for someone who has the board to do it for ease of testing...

enesbcs, 2019-07-06:
I am afraid we have completely different idea of what it means to be "simple". :) I doubt that i will be the "someone", it is way beyond my hardware knowledge, sorry.

cdrose, 2019-07-13:
I need to do a board definition for a 103ZET6 at some point too, I will eventually get round to it but if someone has time to look at the RET6, I would expect the ZET6 to be very similar too (hint hint ;-))

pkourany, 2019-07-25:
I have started work on a 103RCT6 variant but need to figure out the pin mappings for the variant.x files.  The only difference with the RET6 is the flash size.  Hopefully I can finish it in the next week or so if that helps any.

pkourany, 2019-08-06:
I believe I have the 103RCT6 variant working.  My board has no need for Arduino pins as nothing is broken out.  The Blue Button board has almost everything broken out.  This simply affects the way Arduino pin designations are allocated in variant.c and variant.h.  Does anyone care about a specific allocation or I'll simply define them based on left/righ headers and a zig-zag numbering scheme?

enesbcs, 2019-08-06:
I am currently using Roger Clark's Core which simply numbers sequentially the pins from PA0=0,PA1=1 .. PD2=50
https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/master/STM32F1/variants/generic_stm32f103r/board/board.h
But as you wish. :)

pkourany, 2019-08-06:
Yup, that's what I did for my board.  I'll start work in the STM32F103RET6 variant with that approach.

pkourany, 2019-08-07:
Ok, I believe this RET6 variant is good.  I have attached a zip of the variant folder.  Do not rename this folder.  The following needs to be added to the boards.txt file under the `Generic F1` section of the file:
```
# BLUEBUTTON_F103RET6 board
GenF1.menu.pnum.BLUEBUTTON_F103RET6=Blue Button F103RET6
GenF1.menu.pnum.BLUEBUTTON_F103RET6.upload.maximum_size=524288
GenF1.menu.pnum.BLUEBUTTON_F103RET6.upload.maximum_data_size=65536
GenF1.menu.pnum.BLUEBUTTON_F103RET6.build.board=NUCLEO_F103RB
GenF1.menu.pnum.BLUEBUTTON_F103RET6.build.product_line=STM32F103xE
GenF1.menu.pnum.BLUEBUTTON_F103RET6.build.variant=BLUEBUTTON_F103RET6
```

[BLUEBUTTON_F103RET6.zip](https://github.com/stm32duino/Arduino_Core_STM32/files/3477332/BLUEBUTTON_F103RET6.zip)

Let me know how it goes!

enesbcs, 2019-08-07:
Thanks, but when i tried to compile the Blink sample i got the following error:
```
/home/enes/.arduino15/packages/STM32/hardware/stm32/1.6.1/cores/arduino/stm32/timer.c: In function 'TIM8_UP_IRQHandler':
/home/enes/.arduino15/packages/STM32/hardware/stm32/1.6.1/cores/arduino/stm32/timer.c:1328:21: error: 'TIMER13_INDEX' undeclared (first use in this function); did you mean 'TIMER3_INDEX'?
   if (timer_handles[TIMER13_INDEX] != NULL) {
                     ^~~~~~~~~~~~~
                     TIMER3_INDEX
```

pkourany, 2019-08-07:
Yup.  There is a bug in one of the Core files.  See this issue https://github.com/stm32duino/Arduino_Core_STM32/issues/585 for the fix.

enesbcs, 2019-08-07:
OK, now the Blink example compiles without error. 
But after i've uploaded the sketch to the device through Maple DFU Bootloader 2.0, neither the LED blinking starts, nor the DFU port appears. I guess i have to reupload the bootloader to it with a serial bootloader again.

pkourany, 2019-08-07:
I don't use a bootloader so that's as far as I can help.

enesbcs, 2019-08-15:
I've uploaded the compiled Blink sketch and uploaded directly to F103RET6 through serial, but it did not do anything: blinking do not start and there are no data coming from serial... this is the most basic sketch i know, do you have any suggestions for testing?
(I've compiled the same sketch with Roger Clark's core and the same sketch works on the same device through serial upload.)

pkourany, 2019-08-15:
Did you make sure to assign the correct pin for the built-in LED in the varian.h file?

enesbcs, 2019-08-15:
Honestly i never used built in LED definition, i know that my LED is located on PB11 so i am using this:
```
void setup() {
  Serial.begin(115200);
  pinMode(PB11, OUTPUT);
  Serial.print(F("\nSetup"));
}

void loop() {
  Serial.print(F("\nLoop"));
  digitalWrite(PB11, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);                       // wait for a second
  digitalWrite(PB11, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);                       // wait for a second
}
```
This code is working on the other core, led is blinking and "Loop" string arrives from serial, but on this official core, there are nothing happens after uploading&restart. I thought that at least the serial comm has to be working, but nope.

fpistm, 2019-08-19:
> 
> 
> OK, now the Blink example compiles without error.
> But after i've uploaded the sketch to the device through Maple DFU Bootloader 2.0, neither the LED blinking starts, nor the DFU port appears. I guess i have to reupload the bootloader to it with a serial bootloader again.

I dd not check the variant provided by @pkourany but if the ldscript does not take in account the BL_SIZE then after a flash the bootloader is simply overwritten.

pkourany, 2019-08-19:
@fpistm, that is most likely the cause as I am not using a bootloader in my scenario.

fpistm, 2019-08-19:
That's why the ldscript is preprocessed then this is automatically handled.
https://github.com/stm32duino/Arduino_Core_STM32/blob/6d9575ec6a4452d52a2b0d73656c3a060dd0e2c6/variants/PILL_F103XX/ldscript.ld#L55

https://github.com/stm32duino/Arduino_Core_STM32/blob/6d9575ec6a4452d52a2b0d73656c3a060dd0e2c6/variants/PILL_F103XX/ldscript.ld#L63-L64

In platform.txt:
`-Wl,--defsym=LD_FLASH_OFFSET={build.flash_offset} -Wl,--defsym=LD_MAX_SIZE={upload.maximum_size} -Wl,--defsym=LD_MAX_DATA_SIZE={upload.maximum_data_size}`

Depending of the upload method the proper definitions are used:
https://github.com/stm32duino/Arduino_Core_STM32/blob/6d9575ec6a4452d52a2b0d73656c3a060dd0e2c6/boards.txt#L779-L780

fpistm, 2019-08-20:
Could you provide a PR for this then I could review it/help.

fpistm, 2019-08-26:
@pkourany, any update for this issue?

Note link that could be used to reference it in the README.md thanks @ThomasGravekamp work.
https://stm32-base.org/boards/STM32F103RET6-Generic-Board

fpistm, 2019-08-26:
> 
> 
> @fpistm are you looking to me to do the PR as I had not planned to do so?

In fact I thought you will provide it, as I understood you've made the variant ❓

enesbcs, 2019-08-26:
what recommandations?

uzi18, 2019-08-26:
@fpistm maybe it is a good time to provide some generic mcu variants, with more menu options like for esp8266 etc. , it will resolve lots of issues

fpistm, 2019-08-26:
> 
> 
> @fpistm maybe it is a good time to provide some generic mcu variants, with more menu options like for esp8266 etc. , it will resolve lots of issues

any PR's are welcome 😄

fpistm, 2019-08-28:
I've made a PR based on @pkourany variant.
I've fixed some issues and provide 4 boards support:
 * STM32F103R8T6
 * STM32F103RBT6
 * STM32F103RCT6
 * STM32F103RET6

enesbcs, 2019-08-28:
Thank you I've made some tests, it looks like almost perfect, only the HID upload method is failing.
Blink sketch with
- Serial upload: Upload OK, after restart Serial print works, LED blinking OK
- HID 2.2.1 upload: Upload OK, after restart nor serial, neither blinking works
- DFU 2.0 upload: Upload OK, after restart Serial print works, LED blinking OK

fpistm, 2019-08-28:
@enesbcs thanks for the test.

> * HID 2.2.1 upload: Upload OK, after restart nor serial, neither blinking works

The bootloader is for a C8 or CB so this is normal, I guess. @Serasidis do you confirm?

Serasidis, 2019-08-29:
>     * HID 2.2.1 upload: Upload OK, after restart nor serial, neither blinking works

Burn again the HID bootloader and then, select from Arduino IDE:
 `Tools > USB Support (if available) > CDC (generic 'Serial' supersede U(S)ART)`

Upload your Blink sketch through **HID Bootloader 2.2**

fpistm, 2019-08-29:
Thanks @Serasidis

Serasidis, 2019-08-29:
After a second glance, I noticed that @enesbcs  uses the **F103RCT** MCU that is a high density device.
The **F103RCT** has 2k flash sectors instead of 1k flash sectors that have the **F103C8T6** and **F103RBT6** 
That means that the high density F103 devices need different handling during programming.

I will fix that

closed 2019-08-30 by fpistm

Serasidis, 2019-08-30:
@enesbcs Could you please test this bootloader file ?

[hid_bootloader_f103rct.zip](https://github.com/stm32duino/Arduino_Core_STM32/files/3559122/hid_bootloader_f103rct.zip)

Serasidis, 2019-08-30:
STM32 HID Bootloader v2.2.2 (Support for High Density F103 Devices)
https://github.com/Serasidis/STM32_HID_Bootloader/commit/161fb7db0ce93ca7eda6d9f0dfc3024bcc143dc0#diff-04c6e90faac2675aa89e2176d2eec7d8

enesbcs, 2019-08-30:
> @enesbcs Could you please test this bootloader file ?
Sure, i will try it tonight.

enesbcs, 2019-08-30:
Thank you @Serasidis it works flawlessly! All RET owner owe you a coffee! :)
Thanks also to @pkourany and @fpistm for expanding the core with new board support!

PR #615 (2019-08-30): Add Generic STM32F103Rx boards
Four variants available:
 * STM32F103R8T6
 * STM32F103RBT6
 * STM32F103RCT6
 * STM32F103RET6

Based on @pkourany work.

If someone can test them this would be fine as I do not have those boards.
Thanks in advance.

Fixes #556
commit f49877a3 (2019-08-30): Add Generic STM32F103Rx boards

Four variants available:
 * STM32F103R8T6
 * STM32F103RBT6
 * STM32F103RCT6
 * STM32F103RET6

Fixes #556