stm32duino/Arduino_Core_STM32 #558: ADC clock tuning

DmitriyGrigorov, 2019-07-05:
Hi! I use STM32F407VGT6 together with ARDUINO IDE 1.8.9 and Arduino Core STM32 and I have the following question: for my project I need to use the ADC frequency above the standard one, but for this I need to change the divider parameters and sampling rate. I read about the ADC settings in the Arduino Core STM32 wiki, but the result does not change when using these functions, that is, the digitizing time remains the same (60 µs). What am I doing wrong and is it worth considering some additional settings in the sketch for this?

` 
//#include "stm32f4xx_hal_adc.h"
/*
#define ADC_RESOLUTION ADC_RESOLUTION_10B
#define ADC_RANGE 1024
*/
#define ADC_CLOCK_DIV ADC_CLOCK_DIV_4
#define ADC_SAMPLETIME ADC_SAMPLETIME_3CYCLES
void setup() {
  // put your setup code here, to run once:
  analogReadResolution(ADC_RESOLUTION);
  //adc_set_sample_rate(ADC_SAMPLETIME_3CYCLES);
  //ADC1->SMPR2 |= (ADC_SMPR2_SMP1_2 | ADC_SMPR2_SMP1_1 | ADC_SMPR2_SMP1_0); //
Serial.begin(115200);
}

void loop() {
  // put your main code here, to run repeatedly:
  unsigned long t1 = micros ();
  int a = analogRead(PA2);
  
Serial.println(micros()-t1);

}`

fpistm, 2019-07-05:
Hi  @DmitriyGrigorov 
As mentioned in the Wiki:

> ADC_SAMPLINGTIME and ADC_CLOCK_DIV could also be redefined by the variant or using build_opt.h.

It could not be redefined at sketch level as this is not a header.
You should follow this wiki page to redefine them:
https://github.com/stm32duino/wiki/wiki/Customize-build-options-using-build_opt.h

`-DADC_CLOCK_DIV=ADC_CLOCK_DIV_4 -DADC_SAMPLETIME=ADC_SAMPLETIME_3CYCLES`

Note: you should have at least Core  version 1.6.0.

Please, for further support question, use the forum: stm32duino.com.
As it is mentioned here:
https://github.com/stm32duino/Arduino_Core_STM32#troubleshooting

Thanks

closed 2019-07-10 by fpistm

Alteregoxxx, 2019-12-15:
Hello, I have the same need of modifying adc clock divider and sample rate and I got it has to be done using build_opt.h.
Problem is I'm a PlatformIO user and the wiki doesn't say if it is supported/possible and eventually how to do it in PlatformIO.

I also asked in the PlatformIO community, here

[https://community.platformio.org/t/how-to-use-stm32duino-official-stm32-core-build-opt-h-in-platformio/11078](url)

Thank you for your awesome work!

fpistm, 2019-12-16:
@Alteregoxxx 
I could not help on PIO as I never use it.
Anyway it seem @maxgerhardt answer you:
https://community.platformio.org/t/how-to-use-stm32duino-official-stm32-core-build-opt-h-in-platformio/11078/2

Alteregoxxx, 2019-12-16:
@fpistm 
Thank you, @maxgerhardt suggestion worked for using custom build options in PIO; unfortunately, it seems the syntax you suggested,

`-D ADC_CLOCK_DIV=ADC_CLOCK_DIV_4 `

is not right, in fact I receive the following error at compile time, both, in the Arduino IDE and in PIO:

`CCompiling .pio\build\gaito\FrameworkArduino\stm32\bootloader.c.o
:\Users\Gaetano Cirillo\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\analog.cpp: In function 'uint16_t adc_read_value(PinName)':
C:\Users\Gaetano Cirillo\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\analog.cpp:792:18: error: 'struct ADC_InitTypeDef' has no member named 'ClockPrescaler'
   AdcHandle.Init.ClockPrescaler        = ADC_CLOCK_DIV;                 /* (A)synchronous clock mode, input ADC clock divided */
                  ^~~~~~~~~~~~~~
<command-line>:0:15: error: 'ADC_CLOCK_DIV_4' was not declared in this scope
C:\Users\Gaetano Cirillo\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\analog.cpp:792:42: note: in expansion of macro 'ADC_CLOCK_DIV'
   AdcHandle.Init.ClockPrescaler        = ADC_CLOCK_DIV;                 /* (A)synchronous clock mode, input ADC clock divided */
                                          ^~~~~~~~~~~~~
<command-line>:0:15: note: suggested alternative: 'ADC_CLOCK_DIV'
C:\Users\Gaetano Cirillo\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\analog.cpp:792:42: note: in expansion of macro 'ADC_CLOCK_DIV'
   AdcHandle.Init.ClockPrescaler        = ADC_CLOCK_DIV;                 /* (A)synchronous clock mode, input ADC clock divided */`

fpistm, 2019-12-16:
@Alteregoxxx 
Please use the PIO forum.

Alteregoxxx, 2019-12-16:
> @Alteregoxxx
> Please use the PIO forum.

@fpistm 
Frederic, **the compiling error happens with the official Arduino IDE too**, using the indications as per wiki page: creating a build_opt.h containing the following:
`-D ADC_CLOCK_DIV=ADC_CLOCK_DIV_4 
 -D ADC_SAMPLETIME=ADC_SAMPLETIME_3CYCLES`
gives the aforementioned error at compile time.

That's the reason I was asking here, because it seems related to the core, not to the IDE.

fpistm, 2019-12-16:
@Alteregoxxx 

I've already corrected my previous comment. There should be no space between "`-D `" and name
`-DADC_CLOCK_DIV=ADC_CLOCK_DIV_4 -DADC_SAMPLETIME=ADC_SAMPLETIME_3CYCLES`

Note that I did not check if those values exist for F103.

Alteregoxxx, 2019-12-16:
> @Alteregoxxx

> Note that I did not check if those values exist for F103.

I suspect that's the problem: I'm going down in the rabbit hole of your core and it seems that, for the genericSTM32f103Cx mcu, I have to refer to the variant.cpp file, where adc prescaler seems set using the line code

`PeriphClkInit.AdcClockSelection = RCC_ADCPCLK2_DIV6;`

and, from my partial understanding of the whole core design, I fear I can not use the build_opt.h approach for this mcu, because that would mean redefining RCC_ADCPCLK2_DIV6 ...  :-(

fpistm, 2019-12-16:
In that case you Can redefine the `SystemClock_Config(void)`
This is a weak function.