stm32duino/Arduino_Core_STM32 #559: USB library

sstaub, 2019-07-05:
Can you add an USB library like the Mbed library [(https://github.com/ARMmbed/mbed-os/tree/master/usb/device)] ? Using the native USB port would be a great feature.

fpistm, 2019-07-06:
There is already a USB device support based on [STM32_USB_Device_Library](https://github.com/stm32duino/Arduino_Core_STM32/tree/master/system/Middlewares/ST/STM32_USB_Device_Library).

Could you be more precise on what do you expect?

kgamecarter, 2019-07-07:
I think more examples will be fine.
Like examples of maple mini stm32duino.

[image]

sstaub, 2019-07-07:
There are many things I miss, because there is no API like mbed [https://os.mbed.com/docs/mbed-os/v5.13/apis/usb.html](url) :
- Assigning an USBxxx class to USB RX/TX pins (e.g. PA_11/12 for NULEO-L423KC)
- VID/PID settings
- blocking / nonblock
Generell it should possible to work with the build in USB instead of the st-link debugging interface

ahuxtable, 2019-09-03:
No examples on how to use the Keyboard and mouse features, and no joystick support like Roger Clark has.

fpistm, 2019-10-08:
> No examples on how to use the Keyboard and mouse features, and no joystick support like Roger Clark has.

USB keyboard and mouse are functional, enable` HID (keyboard and mouse)` and use Arduino built-in example available in `09. USB` menu.

fpistm, 2019-10-08:
Track in #687

closed 2019-10-08 by fpistm